miguelbalboa/rfid #491: Citations without proper link

grapheo12, 2019-08-26:
In the docs, the citation contains reference numbers which do not have proper links to the original papers.

Rotzbua, 2019-08-26:
@grapheo12 Will you fix it? Or is this just a complain about a 5 year old document which looks like copy&paste?

grapheo12, 2019-08-26:
Surely! Just can you help me find the actual papers?

leopold-gravier, 2024-02-17:
Hello,
Here are replacement links for the broken ones in the documentation.

Ultralight :
https://www.nxp.com/docs/en/data-sheet/MF0ICU1.pdf

Ultralight C :
https://www.nxp.com/docs/en/data-sheet/MF0ICU2_SDS_32.pdf

I cannot do a pull request right now, but I'll try to do it in a couple of weeks.

Have a nice day,
Leopold

leopold-gravier, 2024-03-11:
My bad.
Here's a backup of the wikipedia page that the documentation quotes : http://web.archive.org/web/20140225202826/https://en.wikipedia.org/wiki/MIFARE . The proper links can be found in the Reference tab.
Other thought : It may be wise to say explicitely that the doc is quoting wikipedia