miguelbalboa/rfid #493: Library causing programmer error on Arduino Micro

frederikheld, 2019-10-04:
### Step 1: Describe your environment

 * OS version: Ubuntu 19.04, Gnome 3.21.1, 64-bit
 * Arduino IDE version: 1.8.10
 * MFRC522 Library version: 1.4.5 installed from Library Manager
 * Arduino device: Arduino Micro
 * MFRC522 device: https://funduino.de/nr-18-rfid-kit
  
### Step 2: Describe the problem

As soon as I add the line

```
#include <MFRC522.h>
```

to my sketch, I get programmer errors.

#### Affected file(s) or example(s):

Full example:

```
#include <SPI.h>
#include <MFRC522.h>

// hardware config:
#define RFID_PIN_RESET 7
#define RFID_PIN_SCK SCK
#define RFID_PIN_MISO MISO
#define RFID_PIN_MOSI MOSI
#define RFID_PIN_SS SS

// define RFID:
MFRC522 rfid(RFID_PIN_SS, RFID_PIN_RESET);
MFRC522::MIFARE_Key key;
String byteArrayToString(byte *buffer, byte bufferSize);

void setup() {

  // init heartbeat led:
  pinMode(HEARTBEAT_LED_PIN, OUTPUT);

  // init debug serial (usb):
  Serial.begin(115200);
  Serial.println("");
  
  // init rfid module:
  Serial.println("Initalizing RFID ... ");
  SPI.begin();
  rfid.PCD_Init();
  Serial.println("Done.");
    
  // Indicate "I am ready":
  Serial.println("I'm ready.");

}

void loop() {

  // re-start loop if no new card present:
  if (!rfid.PICC_IsNewCardPresent()) {
    return;
  }

  // re-start loop if reading from card fails:
  if (!rfid.PICC_ReadCardSerial()) {
    Serial.println("read card serial failed");
    return;
  }

  // read UID from card and print it to serial:
  String UID = byteArrayToString(rfid.uid.uidByte, rfid.uid.size);
  Serial.print("RFID UID: ");
  Serial.print(UID);
  Serial.println("");

  // make reader wait for card being removed:
  rfid.PICC_HaltA();

}

String byteArrayToString(byte *buffer, byte bufferSize) {
  String string = "";
  for (byte i = 0; i < bufferSize; i++) {
      string.concat(String(buffer[i] < 0x10 ? " 0" : " "));
      string.concat(String(buffer[i], HEX));
  }
  return string;
}

```
  
#### Steps to reproduce:

  1. Copy example into arduino IDE
  2. Select Arduino/Genuino Micro as Board
  3. Compile & Upload
  
#### Observed Results:

The errors happen during upload but there's different errors on each upload attempt. This is one example that comes very often:

```
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: set addr
avrdude: butterfly_recv(): programmer is not responding
avrdude: verification error, first mismatch at byte 0x0a20
         0x5a != 0x20
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch
```
Others usually revolve around the fact that the serial port is not available.

This one is very common:

```
avrdude: ser_open(): can't open device "/dev/ttyACM0": Permission denied
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```
Before the upload attempt, the port is available but it seems to be reset during upload as it is greyed out or unselected in Arduino IDE after the upload attempt. Sometimes it comes back after some time, usually with a different name (/dev/ttyACM0 --> /dev/ttyACM1). Sometimes it requires a reset to get it back.

Overall it's very erratic behaviour, but it all seems to revolve around the serial port.
  
#### Expected Results:

Upload is working
  
#### Relevant Code:

see above

#### Questions I ask myself:

* Did I choose the right programmer? (I tried with AVRISK mkII and Parallel Programmer but no difference)
* Is my Arduino Micro defective? Is there a way to check that? Without MFRC522.h it's performing well...

Rotzbua, 2019-10-05:
> Did I choose the right programmer? (I tried with AVRISK mkII and Parallel Programmer but no difference)

I think as long it works, it is ok.

> Is my Arduino Micro defective?  Is there a way to check that? 

If it works, no. If you have a clone: sometimes are slightly different parts and wiring is used which may result in other behavior than the genuine.

> Without MFRC522.h it's performing well...

The Micro has no dedicated flashing chip. So it means the chip is used as working and flashing device. This can cause problems. I own a other device where I have to press reset before it can be flashed if it runs longer. Some things may help you: 

* Also the `loop` is a busy one, may add a small delay.
* Wiring can cause this error. 

Closed because not a bug or directly related to library.

closed 2019-10-05 by Rotzbua

frederikheld, 2019-10-06:
Thank you very much @Rotzbua! You set me on the right track by mentioning the specific problems that the flashing setup of the Micro can cause. I dug deeper and apparently the USB serial on my Micro is broken. I got it working by flashing it using an Arduino Uno as programmer :-) Sorry for bothering you, apparently it had nothing to do with this library.