miguelbalboa/rfid #495: PICC_IsCardPresent instead of PICC_IsNewCardPresent?

ElVasquito, 2019-10-15:
Details:

 * OS version: Windows 10
 * Arduino IDE version: Latest
 * MFRC522 Library version: Latest Master
 * Arduino device: ESP8266

Problem:

I call my Read function constantly in my loop, this is it:
`
bool ReadRFID(void) {

  if (!mfrc522.PICC_IsNewCardPresent())
    return false;

  if (!mfrc522.PICC_ReadCardSerial())
    return false;

  Serial.print("Card serial: ");
  for (uint8_t i = 0; i < 4; i++)
    Serial.print(mfrc522.uid.uidByte[i], HEX);
  Serial.println();
  Beep(150, 2); // Beep beep
  return true;
}
`

I have a busy loop, it blocks for about 2-3 seconds almost constantly to check sensors and reply JSON values from server, every few seconds. If a tag is read in those "blocked" seconds, it simply won't read, so it must be taken out and retry (some times a lot) to get the proper reading.
It seems the reader will only read for a second, and then goes off like if the card was removed. The PICC_IsNewCardPresent and PICC_ReadCardSerial functions always return false.
I tried this as well:
`
static bool CardFound = false;

bool ReadRFID(void) {

  if(!CardFound) {
    if (!mfrc522.PICC_IsNewCardPresent()) {
      return false;
    } else {
      CardFound = true;
    }
  }

  if (!mfrc522.PICC_ReadCardSerial()) {
    CardFound = false;
    return false;
  }

  Serial.print("Card serial: ");
  for (uint8_t i = 0; i < 4; i++)
      Serial.print(mfrc522.uid.uidByte[i], HEX);
  Serial.println();
  Beep(150, 2); // Beep beep
  return true;
}
`
But still no luck, ReadCardSerial returns false even after reading the tag properly in the previous loop, I didn't expect that.
So is it possible to have a PICC_IsCardPresent function somehow? I mean, a function that returns true if a card is in the field of the reader (so it can be read anytime) and false if no card is found?
I know this has been asked before, as I found some related issues, and even a promising MP3 player that plays while a tag is in the field and stops when it is removed, but the code did not make any difference to me, I got the same behaviour. Can it be a bad reader?

P.S.: Right before clicking Submit, I got an idea... Could it be the RFM69 radio module I am using, that is sharing the SPI bus with the reader (and constantly receiving data) making some kind of conflict?
It indeed is. I disabled its function and lo a behold, the reader started to work as expected!!
So now the question turned on how to solve the conflict, if at all possible :)

Rotzbua, 2019-10-17:
@ElVasquito Hint: If I remember correctly, there was a similar question in the past. Maybe you find it and can link it here.

ElVasquito, 2019-10-17:
About constant reading, do you refer to #279? Or to #437? That part is already solved anyway.
Now about the SPI conflict that seems to be the issue, I only found a related reference to conflicts with the Ethernet shield (#21), which was of no help in my case.
I checked the RFM69 library I am using (from Felix Rusu) and the mode it is using for SPI is the same (MODE0), SPI_CLOCK_DIV is also 4, and bit order is also MSBFIRST, so all matches your library. I am out of clues.
Side note: I am not connecting the Reset pin of the reader, so it is making a soft reset at setup only, but I don't think that matters in this case either.

tcurdt, 2020-12-02:
Does anyone have an explanation of the alternating behavior? This quick test looks really rather odd.

```
void readCard() { // IRQ
  irqs += 1;
}

void loop() {

  bool gotNewCard = mfrc522.PICC_IsNewCardPresent();
  bool gotSerial = mfrc522.PICC_ReadCardSerial();

  Serial.print(irqs);
  Serial.print(gotNewCard ? " new " : " - ");
  Serial.print(gotSerial ? " serial " : " - ");
  for (uint8_t i = 0; i < 4; i++) {
      Serial.print(mfrc522.uid.uidByte[i], HEX);
  }
  Serial.println();
  irqs = 0;
  
  delay(1000);
}
```

```
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 -  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
  0 -  - E94ABAB1
  2 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
  0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
| 3 new  serial E94ABAB1
| 0 -  - E94ABAB1
```

And if `PICC_IsNewCardPresent` is not called neither does the IRQ get triggered or does `PICC_ReadCardSerial` work.