miguelbalboa/rfid #496: RC522 reader not reading all RFID cards

sslohara, 2019-10-24:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Rotzbua, 2020-04-12:
@sslohara Any update? Otherwise I close this.

closed 2020-04-13 by Rotzbua

karapazar, 2020-08-28:
https://www.eluke.nl/2018/03/08/fixed-rc522-rfid-reader-not-reading-some-cards-part-1/