miguelbalboa/rfid #497: Use with HSPI on esp32

mellowmonk, 2019-11-28:
I have been trying to get the RC522 to work with the esp32 dev board on the HSPI.

After going through forums and discussions, the conclusion for them all is to ask the developer to add the feature.

https://github.com/espressif/arduino-esp32/issues/1219

https://github.com/nkolban/esp32-snippets/issues/61

Maybe it could be a future feature to support HSPI :-D

Thanks

ricardojlrufino, 2020-01-07:
Try using this:
```
#define RFID_SDA 5
#define RFID_SCK 18
#define RFID_MOSI 23
#define RFID_MISO 19
#define RFID_RST 22

 SPI.begin(RFID_SCK, RFID_MISO, RFID_MOSI);
  mfrc522.PCD_Init();
```

fmuntean, 2020-05-14:
> Try using this:
> 
> ```
> #define RFID_SDA 5
> #define RFID_SCK 18
> #define RFID_MOSI 23
> #define RFID_MISO 19
> #define RFID_RST 22
> 
>  SPI.begin(RFID_SCK, RFID_MISO, RFID_MOSI);
>   mfrc522.PCD_Init();
> ```

Actually those are the pins for VSPI which are the default ones used in ESP32
To use the HSPI pins:
 #define HSPI_MISO   12
    #define HSPI_MOSI   13
    #define HSPI_SCLK   14
    #define HSPI_SS     15
      SPI.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI);

gojimmypi, 2020-05-29:
fwiw I was not able to get HSPI working, but VSPI seemed to be fine, following [these instructions](https://www.instructables.com/id/ESP32-With-RFID-Access-Control/). Similar to @fmuntean above, but:

```
// connections, but not #DEFINE
// SDA 21
// SCK 18
// MOSI 23
// MISO 19
// RST 22
```
Actual settings / init:

```
#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN    21
#define RST_PIN   22
#define SIZE_BUFFER     18
#define MAX_SIZE_BLOCK  16
 
MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class

MFRC522::MIFARE_Key key; 
```
Here's a visual on my ThingPulse ESP32 board (IRQ not use):

[image]
[image]

devrim-oguz, 2020-07-15:
There is a pull request on this. Can the library owner accept it, please?

devrim-oguz, 2020-07-15:
I added a new pull request with a smaller solution. Solves this issue.

Rotzbua, 2020-07-25:
Injectable SPI is available at the fork.

closed 2020-07-25 by Rotzbua