miguelbalboa/rfid #503: Deprecated Warnings

jacoblukewood, 2020-01-19:
### Step 1: Describe your environment

 * OS version: macOS 10.15.1
 * Arduino IDE version: 1.8.10
 * MFRC522 Library version: 1.4.5 
 * Arduino device: Mega2560 (not relevant, happens during verify/compile)
 * MFRC522 device: eBay (not relevant, happens during verify/compile)
  
### Step 2: Describe the problem
Compiling any file (including the examples) produces a large number of deprecation warnings that cannot be disabled in any way, polluting the output and hiding useful warnings.
#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. Open any example, for example, ReadNUID.
  2. Verify
  3. View output. Note that the Arduino IDE will only show these on the first compile of each program run. You'll need to quit and verify again to re-see the warnings. Alternatively, an IDE such as VS Code will show these every time.
  
#### Observed Results:

  `In file included from /Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522Extended.cpp:8:0:
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522Extended.h: In constructor 'MFRC522Extended::MFRC522Extended(uint8_t)':
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522Extended.h:81:44: warning: 'MFRC522::MFRC522(byte)' is deprecated: use MFRC522(byte chipSelectPin, byte resetPowerDownPin) [-Wdeprecated-declarations]
  MFRC522Extended(uint8_t rst) : MFRC522(rst) {};
                                            ^
In file included from /Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522Extended.h:10:0,
                 from /Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522Extended.cpp:8:
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.h:339:2: note: declared here
  MFRC522(byte resetPowerDownPin);
  ^~~~~~~
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp: In member function 'bool MFRC522::MIFARE_SetUid(byte*, byte, bool)':
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp:1853:39: warning: 'bool MFRC522::MIFARE_OpenUidBackdoor(bool)' is deprecated: will move to extra class in next version [-Wdeprecated-declarations]
  if (!MIFARE_OpenUidBackdoor(logErrors)) {
                                       ^
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp:1709:6: note: declared here
 bool MFRC522::MIFARE_OpenUidBackdoor(bool logErrors) {
      ^~~~~~~
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp: In member function 'bool MFRC522::MIFARE_UnbrickUidSector(bool)':
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp:1882:34: warning: 'bool MFRC522::MIFARE_OpenUidBackdoor(bool)' is deprecated: will move to extra class in next version [-Wdeprecated-declarations]
  MIFARE_OpenUidBackdoor(logErrors);
                                  ^
/Users/user/Documents/Arduino/libraries/rfid-master/src/MFRC522.cpp:1709:6: note: declared here
 bool MFRC522::MIFARE_OpenUidBackdoor(bool logErrors) {
      ^~~~~~~`
  
#### Expected Results:

  I expect the examples to compile without warnings or at least be able to disable the ones regarding deprecation.

Rotzbua, 2020-01-20:
>  least be able to disable the ones regarding deprecation.

It can be disabled with compiler flags.

> I expect the examples to compile without warnings 

The warnings are deprecation warnings -> warnings for further changes.
As i do not plan to release new versions with functional code changes in the future, the warning can be removed by a pull request.

Rotzbua, 2020-01-20:
@jacoblukewood Can you create a pr?

jacoblukewood, 2020-01-21:
Done

jacoblukewood, 2020-01-21:
> @jacoblukewood Can you create a pr?

Done

Rotzbua, 2020-01-21:
Thanks :+1:

closed 2020-01-21 by Rotzbua