miguelbalboa/rfid #505: Can't compile with "Arduino Nano Every"

dvince38, 2020-01-21:
* OS version: Windows 10 build 1903
 * Arduino IDE version: 1.8.10
 * MFRC522 Library version: v1.4.6
 * Arduino device: Nano Every
 * MFRC522 device: RFID-RC522
  
The MFRC522 Library can't compile with the new "Arduino Nano Every".
It's ok for the old Arduino Nano.

I tried with the different examples provided with the library : Dumpinfo, RFID-cloner, etc.

The error is :

Arduino.h:30:20: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return

 #define F(str) (str)

                    ^

MFRC522.cpp:1341:24: note: in expansion of macro 'F'

   default:      return F("Unknown type");

Rotzbua, 2020-01-21:
That is a problem of the Arduino Nano Every software. Seems they use incompatible types.
Please reports this to arduino team.

closed 2020-01-21 by Rotzbua

KaRa7, 2020-01-22:
Yesterday I was so close to post the exact same issue. Have you already reported it somewhere? Could you keep us updated?

dvince38, 2020-01-23:
I reported the problem to arduino team and there is a patch here :
https://github.com/arduino/ArduinoCore-megaavr/issues/62