miguelbalboa/rfid #508: connecting ttgo  t-display to Rc522

itbetters, 2020-02-26:
* OS version: windows
 * Arduino IDE version:1.8.2
  * Arduino device:ttgo t-display
 * MFRC522 device: SPI rc522 module (the blue one)
  
Hello 
I'm trying to connect the ttgo module to a MFRC522 rfid module . first I decided to use the miguelbalboa library but noticed that there is no hardware SPI pins on the TTgo module (since there are connected to the LCD as I think)
then I found another which I think that would work since we can define software SPI pins there . but when compiling it displays an error for the "application.h"  which is "error compiling for ttgo " .
could you please guide me how I can use rc522 with ttgo board?
regards

Rotzbua, 2020-04-12:
@itbetters Any update? Otherwise I close this.

itbetters, 2020-04-13:
It seems that no one can help me
so please remove
thanks

closed 2020-04-13 by Rotzbua

oguzakilli, 2021-01-01:
> * OS version: windows
> * Arduino IDE version:1.8.2
> * Arduino device:ttgo t-display
> * MFRC522 device: SPI rc522 module (the blue one)
>
> Hello
> I'm trying to connect the ttgo module to a MFRC522 rfid [...]

I think SPI bus is used by display and no way to connect a second device. I'm working on it for days but unfortinately seems impossible.