miguelbalboa/rfid #517: Cannot read  android emulated NFC card

eagleflysky, 2020-05-10:
I'm using your lib to read Mifare Card with ESP8266. It works very well.
However it cannot read  emulated NFC card on my cellphone .
I repace RC522 module with PN532 module and use Adafruit_PN532 lib,  the emulated NFC card can be read out.
I used another cellphone with Taginfo to read the emulated NFC card, it shows below info:
Technologies supported:
  ISO/IEC 14443-3(Type A) compatible
  ISO/IEC 14443-3(Type A) compatible
Android technology information
Tag description:
> TAG:Tech[android.nfc.tech.NfcA]
android.nfc.tech.MifareClassic
android.nfc.tech.NdefFormatable]
android.nfc.tech.NdefFormatable
android.nfc.tech.MifareClassic
android.nfc.tech.NfcA
>Maximum transceive length:253 bytes
>Default maximum transceive time-out:618ms
Detailed protocal information
ID: 3F:B8:88:6F
ATQA:0x0400
SAK:0x08
ATS:0x05788070020000000000000000000000
>Max.accepted frame size:256 bytes(FSCI:8)
>Supported receive rates:
.106kbit/s(DR:1)
>Supported send rates:
.106kbit/s(DR:1)
>SFGT: 302.0us(SFGI:0)
>FWT:38.66ms(FWI:7)
>NAD not supported
>CID supported
>Historical bytes:[none]
Will this lib support  emulated NFC card later? Thanks

Rotzbua, 2020-05-10:
As far as I know, this is a hardware limitation by the mfrc522, as mentioned in the readme. The pn532 work because it has more abilities.

You can cross check with a raspberry + python mfrc522 library. If mfrc522+raspberry+emulated card work, then we have a software problem in this library.

Best.

eagleflysky, 2020-05-13:
Hi,Rotzbua,
    I've tried on Raspi with MFRC522 lib(https://github.com/mxgxw/MFRC522-python.git), and the emulated card can be read out:
>pi@raspberrypi:~/MFRC522-python $ python3 Read.py
Welcome to the MFRC522 data read example
Press Ctrl-C to stop.
Card detected
Card read UID: 51,32,211,4
Size: 8
Sector 8 [3, 17, 17, 20, 255, 255, 255, 25, 18, 18, 0, 0, 255, 255, 3, 16]

Rotzbua, 2020-05-14:
The init config is different in python:

https://github.com/mxgxw/MFRC522-python/blob/cf7f9ec0cc928883700e7153529b50e5fc362ab0/MFRC522.py#L409-L412

Can you replace setting in `MFRC522.cpp`:

https://github.com/miguelbalboa/rfid/blob/ed5c081a54635d41147a47505638a44a663dcfbf/src/MFRC522.cpp#L229-L232

With:

```c++
	PCD_WriteRegister(TModeReg, 0x8D);
	PCD_WriteRegister(TPrescalerReg, 0x3E);
	PCD_WriteRegister(TReloadRegH, 0);
	PCD_WriteRegister(TReloadRegL, 30);
```

Maybe this helps..

eagleflysky, 2020-05-15:
I did the math for the init parameters, it turns out the python library set the reload timer as 24ms before time out. It's not a big difference.

eagleflysky, 2020-05-15:
Anyway , I've tested with modified init, still failed to read the emulated card

eagleflysky, 2020-05-18:
I did 2 more test:
1. Installed micropython(http://micropython.org/resources/firmware/esp8266-20191220-v1.12.bin), modified SPI driver of MFRC522 lib(https://github.com/mxgxw/MFRC522-python.git) for esp8266. Then run the read test, the cellphone can be readout:
>
WebREPL connected                                                                                                                                     
>>> import os                                                                                                                                         
>>> os.listdir()                                                                                                                                      
['boot.py', 'webrepl_cfg.py', 'MFRC522.py', 'reading.py']                                                                                             
>>> import reading                                                                                                                                    
>>> reading.reading()                                                                                                                                 
Card detected - 16                                                                                                                                    
Card read UID: 51,32,211,4                                                                                                                            
Size: 8                                                                                                                                               
Sector 8 [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

eagleflysky, 2020-05-18:
2.  ported MFRC522-python to Arduino, it didn't work -- still failed to read cellphone , it worked for real card

eagleflysky, 2020-06-02:
@Rotzbua Cellphone emulated card can be read after I updated this lib to latest version: )

closed 2020-06-02 by eagleflysky