miguelbalboa/rfid #518: important info missing in the comments

StefanL38, 2020-05-26:
Hello,

after a lot of research I finally managed to make an RC522-reader work together with a ESP8266-Board. The very short but very important information to do this is

1.) Adapt the IO-Pins to the **HARDWARE**-SPI-interface of your processor.
2.) Make sure the IO-Pin connected to Reset has a 10k pullup-resistor.

There is some information about how to connect to Arduino-Boards in the comments of the library
but there is **NO** word that it has to be the hardware-SPI-interface.
This info should be added to the comments at a prominent and easy to find place.
It would have saved me hours of tests and days of waiting for answers in forums.

best regards  

Stefan

Rotzbua, 2020-05-26:
First of all: this is open source, if you want it fixed, create a pull request.

> There is some information about how to connect to Arduino-Boards in the comments of the library
but there is NO word that it has to be the hardware-SPI-interface.

In the world of embedded systems you can not connect stuff as you want. Only specific pins have specific functions.
This library is an Arduino library, so based on the Arduino framework which has a simple `use the first spi interface on the chip` policy... do not blame the library for this.

>  The meaning of the word "typical" is completely different from[..]

Typical = for normal maker connecting reader with arduino

>  2.) Make sure the IO-Pin connected to Reset has a 10k pullup-resistor.

Why? And there are missing information: voltage used, which esp board, exact wiring.

StefanL38, 2020-05-26:
Hi Rotzbua,

i tried to open a pull-request:
clicking the pullrequest-tab
then clicking on the green button named "new pull request"
but the underlying ur l is https://github.com/miguelbalboa/rfid/compare

so the browser showed me a page
Compare changes
Compare changes across branches, commits, tags, and more below. If you need to, you can also .

where the create new pull-request-button is disabled.

How can I activate this button create new pull-request?

best regards  

Stefan

gojimmypi, 2020-05-29:
@StefanL38 as it seems you are new to GitHub, with only 1 repo for your first "Hello World"... I'm sure this is a bit intimidating. Check out the [Creating a Pull Request](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request).

Basically: you need to fork this repo into your account, make some changes, then create the PR from your account, proposing to push changes back here.

StefanL38, 2020-05-30:
OK I understand. I also can see the positive effect to keep the masterbranch clean and working. 
Though my opinion is: creating a fork just for each tiny small thing leads to a "forky jungle". That's not my way of trying to support things.  To me forking makes sense for a substantial change but not  for every small thing. If the majority of users likes this - do it. I'm not in the position for changing it.

gojimmypi, 2020-05-31:
yes, that's kinda just how GitHub works. Don't let that discourage you from contributing. It is my understanding that the backend database is fairly efficient in the way files are stored with hashes and pointers, even for forks. In fact, even the web browser edit will fork - then upon completion offer to delete the fork/branch.

Rotzbua, 2020-05-31:
@gojimmypi  That is not a problem. Even for tiny changes it is ok to create a fork.

Rotzbua, 2020-06-03:
>  2.) Make sure the IO-Pin connected to Reset has a 10k pullup-resistor.

@StefanL38 I think that is not required. I checked two of my mfrc522 modules. Both have a pullup from rst to vcc on the pcb.

closed 2020-06-06 by Rotzbua

gojimmypi, 2020-06-08:
@StefanL38 what does your board look like? Is it the red version?

I have the blue version of the `RFID-RC522` boards, and although many of the schematics I've found show no pull-up on the `RST` line, it appears that my boards have the 1K R2 to Vcc. (and _not_ a pull-up on SDA/Rx as shown on the schematic that I have).

Here's a picture of the board I have:

[image]

and if you look very closely, you can see the R2 pullup connected to `RST`:

[image]

@Rotzbua  Would you happen to have an accurate schematic?

Rotzbua, 2020-06-09:
@gojimmypi I think the schematic from sunrom is close to the most boards. IRQ wiring seems not present in the schematic: https://www.sunrom.com/p/rfid-readerwriter-1356mhz-rc522-spi-module-with-cardkeychain

gojimmypi, 2020-06-09:
@Rotzbua That PCB is indeed very close, but that's not quite the schematic, even for the PCB shown on that page. The first thing I noticed, the IRQ is not connected as you said; the board I have does appear to be using that line.  (although I've not tried actually using it)  The other thing from the schematic vs the PCB are the capacitor labels don't match. However that schematic seems more accurate than the one I have, and does appear to properly show the resistor tying RST high (although it should be R2 and not R14) and no resistor on SDA, as appears to be the case on my board.

I appreciate you sharing that. Thank you.