miguelbalboa/rfid #523: A series of compiler errors are thrown when using this library with ATtiny804 device

TomWS1, 2020-08-05:
### Step 1: Describe your environment

 * OS version: Windows 7
 * Arduino IDE version: 1.8.8
 * MFRC522 Library version: V1.4.6 
 * Arduino device: ATtiny804
 * MFRC522 device: MFRC522
  
### Step 2: Describe the problem
The compiler throws errors when compiling for a board using ATtiny804 device and the megaTinyCore.  The issue is that the boards package redefines __FlashStringHelper as char due to the fact that there is no helper function required to access program memory as data with this series of processors.

This can be fixed with the 'hack' of renaming __FlashStringHelper in MFRC522.h and MFRC522.cpp to MEM_REF_HELPER and adding to MFRC522.h:

#if defined(__AVR_ATtinyx04__)
  #define MEM_REF_HELPER char
#else
  #define MEM_REF_HELPER __FlashStringHelper
#endif

#### Affected file(s) or example(s):
  
  * errors.txt
  * MFRC522.h (with 'hack' fix)
  * MFRC522.cpp (with 'hack' fix)
  
#### Steps to reproduce:

  1. Select Board "ATtiny1614/1604/...", Chip: ATtiny804

  2. Compile any code using the MFRC522 library
  
#### Observed Results:

  * What happened?  See errors.txt
  
#### Expected Results:

  * Compile without errors.
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```
[errors.txt](https://github.com/miguelbalboa/rfid/files/5030397/errors.txt)

[src.zip](https://github.com/miguelbalboa/rfid/files/5030402/src.zip)

Rotzbua, 2020-08-05:
Must be fixed by the framework of the board.
The simple solution was to typedef `__FlashStringHelper` as `char` and define `F()` macro which does not modify the string. But then the RAM consumption is high.

closed 2020-08-05 by Rotzbua

TomWS1, 2020-08-05:
Not quite as simple as that as all functions that attempt to overload with ___FlashStringHelper then fail as duplicate overloads.  I suppose one way around it is to implement __FlashStringHelper so that it beningly returns a char*

Rotzbua, 2020-08-05:
@TomWS1 `Arduino.h` of megatinycore 2.0.5 does redefine `F()` with type `char`:
```cpp
#undef F
#define F(str) (str)
```
`String.h` has a different definition with type `__FlashStringHelper`.

So basically the framework has a problem with different behavior of a macro.
That problem must be solved on framework site.