miguelbalboa/rfid #524: Adding functions for easy access to CardID

StefanL38, 2020-09-06:
Hi everybody,

I'm still not sure if this is the right place to make suggestions. 
The tab pullrequest is talking about "how to fix an issue"

A suggestion for adding new functions is not "fixing an issue"
and it is not an issue itself.
So I will just write it here:

The MFRC522  has functions to print out every single byte to serial 
But it has no functions that would give easy access to the UID and similar things.
Nor does the examples show how you extract it from the whole datastructure.

So my suggestion is to add such functions or pointing me to another version of this library that does have such functions

as a general remark: GitHUB seems to be extremly blind for beginners difficulties to explain how to 
- asking questions about the functionality of something
- suggest enhancements as a pure word-posting 

Of course I can go crawling through the whole website
trying to find it. 

this big "Read the guide" button tries to force me to "contribute" by creating my first repository. 

Ey guys I'm five steps BEFORE creating my first repository that does make sense.
I just want to know where is the right place for posting a suggestion?
 
best regards Stefan

Rotzbua, 2020-09-07:
I have often written the idea of a simple API. But it takes a lot of time to create and maintain it.
Maybe there are already parts of your idea only not yet as a published library but only as a snippet or small blog post.
It would be a pleasure for me to see this idea as a project, but I do not have time for it.

drws, 2021-05-30:
[This](https://stackoverflow.com/questions/32839396/how-to-get-the-uid-of-rfid-in-arduino) is an useful link also including some code examples for the functionality. In principle, the library already reads and processes the UID of the card, only a simple wrapper function to return the UID is left to be desired. Or the uid variable could be made public?