miguelbalboa/rfid #535: 5 RFID Readers for Model Railway to slow

Phil997, 2020-11-07:
### Step 0: Are you in the right place?

Hello

I would like to use MRFC522 to perform a position detection for my model railroad. Thereby the trains as well as the self-propelled cars (Car System) shall be detected. The coil of the token should be positioned at the trains or under the cars. The reader is then located under the road surface.

### Step 1: Describe your environment

 * OS version: Win10
 * Arduino IDE version: 1.8.3
 * MFRC522 Library version: 1.4.7 Master
 * Arduino device: Arduino Nnao
 * MFRC522 device:RC522
  
### Step 2: Describe the problem

With a reader it works well. The car drives over the reader and my Arduino Nano detects the car. However, if I have 5 readers distributed on the model railroad, the vehicles are not always recognized. 

The problem is that only one reader seems to be active at a time and it can recognize a vehicle. The vehicles are so fast that they drive over a reader, but the reader is not activated at this moment and the vehicle is not recognized.

Since the Nano has only 2 interrupts, I can exclude this solution.

Can I use multithreding?
  
  
#### Relevant Code:

```

/*
Arduino Nano
AtmelA328P Old Bootloader

/*
 * Typical pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS 1    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required **
 * SPI SS 2    SDA(SS)      ** custom, take a unused pin, only HIGH/LOW required **
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
 *
 */

#include <SPI.h>
#include <MFRC522.h>

// PIN Numbers : RESET + SDAs
#define RST_PIN         9
#define SS_1_PIN        10
#define SS_2_PIN        8
#define SS_3_PIN        7
#define SS_4_PIN        6
#define SS_5_PIN        5

#define NR_OF_READERS   5
byte ssPins[] = {SS_1_PIN, SS_2_PIN, SS_3_PIN, SS_4_PIN, SS_5_PIN};
MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.

/**
 * Helper routine to dump a byte array as hex values to Serial.
 */
void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}

void setup() {

  Serial.begin(9600); // Initialize serial communications with the PC
  while (!Serial);    // Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
                      // TODO: Remove while loop if nessesary

  SPI.begin();        // Init SPI bus

  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    Serial.print(F("Reader "));
    Serial.print(reader);
    Serial.print(F(": "));
    mfrc522[reader].PCD_DumpVersionToSerial();
  }
}

void readUid(uint8_t readerNumber ){
  Serial.print(readerNumber);
  if (mfrc522[readerNumber].PICC_IsNewCardPresent() && mfrc522[readerNumber].PICC_ReadCardSerial()) {      
    Serial.print(F("readerNumber "));
    Serial.print(readerNumber);
    // Show some details of the PICC (that is: the tag/card)
    Serial.print(F(": Card UID:"));
    for (byte i = 0; i < mfrc522[readerNumber].uid.size; i++){
      Serial.print(mfrc522[readerNumber].uid.uidByte[i], HEX); 
      Serial.print(" "); 
    }
    Serial.println();
  }   
}
 
void loop() {
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    readUid(reader);
  } 
}

```

Rotzbua, 2020-11-07:
Performance issues are ugly... Is it even possible to detect the tag on the train with only one reader reliable?

Phil997, 2020-11-07:
Is difficult. The problem is that there are several ways for the vehicles and trains to be queried. For example for crossings or railswitches to avoid possible collisions

Rotzbua, 2021-01-03:
@Phil997 Any update? Did you tried the new 1.4.8 which sets the spi clock speed to an higher speed?

closed 2021-01-11 by Rotzbua