miguelbalboa/rfid #537: FM17522 Documentation

a-v-s, 2020-11-28:
My apologies, this is not a bug report, but rather a request for information.

I see in the source code you have defined a hardware ID and a firmware reference for the Fudan Micro FM17522. 
I am trying to locate the documentation those values are taken from. I am only able to locate the short datasheet, but no detailed datasheet. Is there any such information, or are these values collected from reading back a chip?

Rotzbua, 2020-11-28:
Reference pr #106.
Maybe the values are from the chip rather than from a data sheet. You could ask @socram8888.

socram8888, 2020-11-28:
The values are from a known working chip indeed. I could not find any datasheet where those would be documented.

Rotzbua, 2020-11-28:
@a-v-s The company is still in business. You could try to request a data sheet.

a-v-s, 2020-11-30:
Thank you for your responses. I'll contact them and see if they reply.

a-v-s, 2020-12-22:
I've tried to contact them, but I've got no response. I wasn't even sure if the contact form on their website was functional.

Nevertheless, I've been looking at their short datasheets for the FM17522 and FM17550. One thing I noticed looking at those datasheet, it mentions the FM17522 supports ISO 14443-A, and the FM17550 both A and B, but it does not mention MIFARE. This makes me wonder whether they are equivalent of MFRC522 and MFRC523, or lack the MFAuthent command.

Rotzbua, 2020-12-23:
Ok, thanks for the feedback.

The FM17522 should be fully compatible to the original mfrc522. So it should support the MFAuthent  command. In reality it looks like  99% of the project only use the uid so it could be that there is a difference or problem.

I will close the issue due the fact, that even the manufacturer does not react or (in my opinion) no full datasheet exists.

closed 2020-12-23 by Rotzbua