miguelbalboa/rfid #540: RFID stops reading after some hours

Magerquark, 2020-12-03:
* Version 1.4.7
 * Arduino device:  Nano 33 IOT
 * MFRC522 device: Joy IT  RFID Reader
  
### Step 2: Describe the problem

The RFID Reader stops reading cards after some hours the program I use is still working  but mfrc522.PICC_IsNewCardPresent() and mfrc522.PICC_ReadCardSerial() both return false even if a new card is put on the reader. 
  
#### Steps to reproduce:

Let the a simple RFID reader programm run for 24-48 hours -> reader will probably stop reading cards.

I need my program to run more than a few hours I think the problem can be fixed by setting the RST PIN to low and PCD_Init(); if the problem occours. 

The problem is I dont exactly know why and **when** this problem occours.
Is there a way from the software side to find out if the reader is still working without putting on some cards manually?

Also I think that the hard reset in PCD_Init(); is not working properly I have to set the RST Pin to low manually before it calling this function because else the reader does not read card after compiling sometimes.

Rotzbua, 2020-12-05:
Try to remove the wired reset physically and in code.

> Also I think that the hard reset in PCD_Init(); is not working properly I have to set the RST Pin to low manually before calling this function because else the reader does not read cards after compiling sometimes.

Could be because the nano33 uses a different chip. The init/behavior of the gpios could be different

Magerquark, 2020-12-05:
"Try to remove the wired reset physically and in code."
Hi thanks for your anwser I am not quite sure what you mean could explain it more? (my english is not very good tho)
At the moment I am not resetting anything?

The dirty workaround would be my last try.
It would be better to just reset the mfrc if it doesnt work anymore is there a way to find out if the mfrc doesnt work anymore?

Rotzbua, 2020-12-06:
> 
> 
> "Try to remove the wired reset physically and in code."
> Hi thanks for your anwser I am not quite sure what you mean could explain it more? (my english is not very good tho)

* Disconnect `RST/Reset` from mfrc522 module.
* Change code

from:

> MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

to

> MFRC522 mfrc522(SS_PIN, UNUSED_PIN);  // Create MFRC522 instance

> At the moment I am not resetting anything?

The library tries to be user friendly and resets the mfrc522 at initialization phase automatically.
Background: The mfrc522 can be reset by electrical signal over RST pin or programatically over SPI.

Magerquark, 2020-12-06:
Thank you for your anwser.
I will try this out and give feedback in the next days, but do you really think that removing the reset at the beginning wil fix that the reader stops working?

Rotzbua, 2021-01-03:
@Magerquark Any update? Did you tried the new 1.4.8 which sets the spi clock speed to an higher speed?

Magerquark, 2021-01-03:
Hi I used to hard and software the reader every 30 seconds for me this seems to work, I tried it for about 4 days and the reader were still running. 
I can try one reader without the reset and the new Firmware if you want.

Rotzbua, 2021-01-04:
> I can try one reader without the reset and the new Firmware if you want.

That would be great 👍

Rotzbua, 2021-01-05:
Quote  similar issue from @wesee-dev 

> ### Step 1: Describe your environment
> 
>     * OS version: Windows 10 Build 2004
> 
>     * Arduino IDE version: 1.8.10
> 
>     * MFRC522 Library version:  1.4.8 master branch
> 
>     * Arduino device: ESP32 WROOM
> 
>     * MFRC522 device: RC522
> 
> 
> ### Step 2: Describe the problem
> 
> The RC522 module stops reading cards after running for a long time, about two weeks.
> After pressing the reset button of the ESP32 module, the module started reading cards again.
> Is there a way to detect if module is unresponsive, or identify what caused this issue by reading a register in the RC522?
> I think that periodically reset the mfrc522 could fix the problem, but I still have not tested this approach.
> #### Affected file(s) or example(s):
> 
> DumpInfo.ino
> #### Steps to reproduce:
> 
>     1. Let the dumpInfo exemple running for a long period, in my case two weeks.
> 
>     2. Test if module is still reading cards.
> 
> 
> #### Observed Results:
> 
> No cards were being read, no output in serial monitor when cards were aproximated.
> The module started reading cards again when the ESP32 board was resetted.
> 
> I already took a look in the issues section but wasn't able to find any similiar issues, but I think this might be a common issue.

For long term debugging a setup with pc and serial output is not the best approach. The usb ports on the pc have several power saving mechanism which can affect the test bench. Furthermore the serial communication can be a problem. I would recommend a dedicated usb power source and an independent indicator for successful reading like a simple led. Best.

jundiprima, 2021-01-26:
> Hi I used to hard and software reset the reader every 30 seconds, for me this seems to work. I tried it for about 4 days and the reader was still running.
> 
> I can try one reader without the reset and the new Firmware if you want.

hi, can you please give more update please, that would be great. thanks

Magerquark, 2021-01-26:
Hey today I set up on reader without the reset and the newest firmware. I will give updates in a few days :)

Magerquark, 2021-02-16:
Hey the reader I set up worked after one week so I think this problem is kind of fixed or need longer time till it stops working.
After 2 weeks my Arduino programm stopped working (not the reader) so I dont know if it works for a longer period of time.
If I find time I can try to set it up again.

Kindly regards
Alexander

Rotzbua, 2021-02-16:
@Magerquark Thanks for the work and feedback 👍. I think a week of successful testing is a good proof. 
I close this now. Can be reopened if somebody finds the cause.

closed 2021-02-16 by Rotzbua

licornelibre2, 2022-05-25:
The problem is that the rfid has stopped working after a period of time. Can you help with a code?

licornelibre2, 2022-05-25:
#define BLYNK_PRINT Serial
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <SPI.h>
#include <MFRC522.h>
#include <EEPROM.h>

#define SS_PIN 4   //D2
#define RST_PIN 5   //D1
#define BTN_PIN 0  //D3
#define SLN_PIN 2  //D4

MFRC522 mfrc522(SS_PIN, RST_PIN);
unsigned long uidDec, uidDecTemp;
int ARRAYindexUIDcard;
int EEPROMstartAddr;
long adminID = 1122539531;
bool beginCard = 0;
bool addCard = 1;
bool skipCard = 0;
int LockSwitch;
unsigned long CardUIDeEPROMread[] = {0, 1, 2, 3, 4, 5, 6, 7, 8, 9};
int PiezoPin = 15;  //D8

char auth[] = "NF_xxx-yyy-zzz_ErFyB";
char ssid[] = "IoT RFID Door Lock";
char pass[] = "RFID@DoorLock";

WidgetLCD lcd(V0);
BlynkTimer timer;

 
void setup() {
  Serial.begin(115200);
  pinMode(SLN_PIN, OUTPUT); digitalWrite(SLN_PIN, LOW);
  pinMode(BTN_PIN, INPUT_PULLUP);
  pinMode(PiezoPin, OUTPUT);

  SPI.begin();
  mfrc522.PCD_Init();

wesee-dev, 2022-05-25:
> #define BLYNK_PRINT Serial #include <ESP8266WiFi.h> #include <BlynkSimpleEsp8266.h> #include <SPI.h> #include <MFRC522.h> #include <EEPROM.h>
>
> #define SS_PIN 4 //D2 #define RST_PIN 5 //D1 #define [...]

Try to reset the module periodically via the RST pin. This solved this issue for me. You can reset the module every hour or so.
Remember to re-initialize the device after reset.
The library also provides methods to soft and hard reset the module via its registers, you can try this first.

ScreamZ, 2023-11-13:
@console-netsphere @Rotzbua 

I've been experiencing some related bugs but the device was freezing like every 10 minutes.

I've removed the RST pin and used `MFRC522 rfid(SS_PIN, MFRC522::UNUSED_PIN);` seems to work. But for me it's like magic, am I supposed to do a manual reset like using a pin on reset and sometimes dispatch a HIGH/LOW signal of 500ms. And then call `mfrc522.PCD_Init();` again? Or could this work this way without anything?

Is there any documentation on that or some feedback on usage?

Also, I can see that there is a https://github.com/OSSLibraries/Arduino_MFRC522v2 which seems to do Software reset