miguelbalboa/rfid #543: RFID Write embeds line feed character

cgeshan, 2020-12-16:
Using Arduino Mega, MIFARE 1KB tag. 

The problem I am having is when using the RFID write example code from the library the code successfully writes the correct block, but it adds a line feed character (0A) to the beginning of the block. I do not wish for it to do this, but I am unaware of how to change this. I would prefer it to simply be a space (20) instead.

Rotzbua, 2020-12-17:
Which exact example?

cgeshan, 2020-12-18:
> Which exact example?

rfid_write_personal_data

Rotzbua, 2020-12-18:
Works as intended.
```c++
  Serial.println(F("Type Family name, ending with #"));
  len = Serial.readBytesUntil('#', (char *) buffer, 30) ; // read family name from serial
```
As you can see, the code has as end the `#` and not a line ending. So if you send by console `#` and press the return button you send two signs. The function copies until the `#` and the return button sign (depending on settings) a line feed character. The line feed character is buffered, so it appears on your next input.

The behavior is indeed not intuitive. Flush/clearing the buffer could solve the problem, but I think there was no function in the arduino api defined https://github.com/arduino/ArduinoCore-API/blob/fc0c12d7faf43782fd44d2e1384abfef6109da21/api/Stream.h . Or as second solution replace the `#` by line feed char (but maybe incompatible on different clients).

So far example work. Feel free to create a pull request with an tested better solution or improved documentation.

closed 2020-12-18 by Rotzbua