miguelbalboa/rfid #544: MFRC522.h:22:26: error: 'SPI_CLOCK_DIV4' was not declared in this scope

maximus1127, 2020-12-28:
### Step 1: Describe your environment

 * OS version: Windows 10
 * Arduino IDE version: 1.8.12
 * MFRC522 Library version: 1.4.7 
 * Arduino device: Nano 33 BLE Sense
 * MFRC522 device: RFID-RC522
  
### Step 2: When loading the library and selecting an example sketch, I'm getting the error of "MFRC522.h:22:26: error: 'SPI_CLOCK_DIV4' was not declared in this scope". Everything is completely stock, the library was downloaded and imported straight into the editor, the example was selected without editing any code or anything. I can't even upload it to the nano as the compile failed with this error. 

  
#### Steps to reproduce:

  1. Import library
  2. Select example sketch "read NUID"
  3. compile and wait for pretty red text

Rotzbua, 2020-12-28:
`SPI_CLOCK_DIV4` should be defined by the board.
What version and board implementation do you use for the Nano 33 BLE Sense ? Look it up at the board manager in the arduino ide.
Thanks

maximus1127, 2020-12-28:
[image]

I'm using the "Arduino Nano 33 BLE" board

Rotzbua, 2020-12-28:
## Solution

Set compiler flag to overwrite with your values according to your board: `MFRC522_SPICLOCK `

## Simple Workaround

* Goto your local copy and change line:
https://github.com/miguelbalboa/rfid/blob/ea10e8a21963cb603167892cce37eab40f7a1bc6/src/MFRC522.h#L22
* Replace `SPI_CLOCK_DIV4` with `0`

## Longterm Solution

Would be to add the constants as compatibility to the board library.

closed 2020-12-28 by Rotzbua

Rotzbua, 2020-12-30:
Solved by 34086ad99a86ff381dc3c4fd2454a6b7b8c3b78d