miguelbalboa/rfid #555: RFID reader is delaying millis() timer less when running more logic and peripherals

martinius96, 2021-03-14:
### Step 1: Describe your environment

 * OS version: Windows 10
 * Arduino IDE version: 1.8.4
 * MFRC522 Library version: Latest release 1.4.8
 * Arduino device: Arduino Uno R3 (CH340)
 * MFRC522 device: NXP RC522 reader with 0x12 (counterfeit chip) and second with 0x92 address (Firmware 2.0)
  
### Step 2: Describe the problem
Hello there.
I have some problem with my sketch that is using RFID reader RC522.
I have two RFID readers RC522, one have address 0x12 (counterfeit chip) and other is 0x92 (Firmware v2.0 with original NXP chip i think), problem is same at both readers (using only one at project). 

Whole project is like STOPWATCH that is couting time each second (RFID card is used for user entry for start, stop or eraser time). In program implementation I am using simple timer... if millis - my_saved_millis() is more or equal to 1000, run logic. I am using also I2C LCD display there. So, fist time is stopped and it is not overwriten on display. In that moment I have running of logic each +- 1030ms (so 30 ms over 1000 ms that should trigger that timer). When I attach card that will start running logic of stopwatch, it is also sending characters to display, but delay is now only 4ms over 1000 ms (1004 ms). 

**So it is so weird, because delay is less when I am using more peripherals...**
I am using more peripherals (RFID and display) and delay is 4ms over 1000ms and if there I am using at the beginning RFID only (before display is overwriten - waiting for first attach of card) I have 30 ms over 1000ms. I have look into .cpp files of MFRC522 library, but I havent found there anything that can cause that problem. Also I tried to set SCK from 4MHz (used in lib) to 8MHz and delay is 2ms over 1000. **Delays over 1000ms are same at COUNTERFEIT or at ORIGINAL chip of NXP RC522 reader**
Printing on LCD display is almost 1ms for 2 characters, there is 16 characters, so it is 8ms per its print on display. I am using standard I2C speed 100KHz.

Have somebody idea what can cause this? Why I have bigger delays when I am nothing printing on display and when counter starts with running full logic, delay is less? That doesn't make sense to me. Thanks, Martin
# UART output
[image]
# Minimal reproductive example
* Edit there NUMBER of CARD (UID), it will print it to UART too (9600 baud/s) when you attach it to RFID reader
* Edit I2C address of display 0x3F used. I2C library is from: https://github.com/fdebrabander/Arduino-LiquidCrystal-I2C-library

#include <LiquidCrystal_I2C.h>
#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN         9          // Configurable, see typical pin layout above
#define SS_1_PIN        10         // Configurable, take a unused pin, only HIGH/LOW required, must be diffrent to SS 2
#define NR_OF_READERS   1
byte ssPins[] = {SS_1_PIN};
MFRC522 mfrc522[NR_OF_READERS];   // Create MFRC522 instance.
LiquidCrystal_I2C lcd(0x3F, 16, 2); //I2C address, DIGITS, ROWS 0x3F is also most used address for LCD display

unsigned long code; //adresa karty
boolean run = false; //bezi prvy cas?
boolean run2 = false; //bezi druhy cas?
long timer = 0; //timer systemu
int second = 0; //sekundy prveho timu
int minute = 0; //minuty prveho timu
int tenth = 0;  //stotiny prveho timu
int hour = 0;   //hodiny prveho timu
int second2 = 0; //sekundy druheho timu
int minute2 = 0; //minuty druheho timu
int tenth2 = 0; //stotiny druheho timu
int hour2 = 0; //hodiny druheho timu
const int rfrele = 3; //LED PRE RF TIM
const int gdprele = 4; //LED PRE GDP TIM
void setup() {
  Serial.begin(9600);
  SPI.begin();
  pinMode(rfrele, OUTPUT);
  pinMode(gdprele, OUTPUT);
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    mfrc522[reader].PCD_Init(ssPins[reader], RST_PIN); // Init each MFRC522 card
    Serial.print(F("Reader "));
    Serial.print(reader);
    Serial.print(F(": "));
    mfrc522[reader].PCD_DumpVersionToSerial();
  }

  lcd.begin();
  lcd.backlight();
  lcd.setCursor(0, 0);
  lcd.print("RF  00:00:00");
  lcd.setCursor(0, 1);
  lcd.print("GDP 00:00:00");
}

void tickClock() {
  if ((millis() - timer) >= 1000 || timer == 0) {
    timer = millis();
    Serial.println(timer);
    tick();
    tick2();
  }

}

void loop() {
  for (uint8_t reader = 0; reader < NR_OF_READERS; reader++) {
    if (mfrc522[reader].PICC_IsNewCardPresent() && mfrc522[reader].PICC_ReadCardSerial()) {
      Serial.print(F("Reader "));
      Serial.print(reader);
      // Show some details of the PICC (that is: the tag/card)
      Serial.print(F(": Card UID:"));
      dump_byte_array(mfrc522[reader].uid.uidByte, mfrc522[reader].uid.size);
      Serial.println();
      Serial.print(F("PICC type: "));
      MFRC522::PICC_Type piccType = mfrc522[reader].PICC_GetType(mfrc522[reader].uid.sak);
      Serial.println(mfrc522[reader].PICC_GetTypeName(piccType));

      // Halt PICC
      mfrc522[reader].PICC_HaltA();
      // Stop encryption on PCD
      mfrc522[reader].PCD_StopCrypto1();
    }
  }

  if (code == 11966 || code == 12354) { //RF TEAM ---> START COUNTING ITS TIME
    digitalWrite(rfrele, LOW);
    digitalWrite(gdprele, HIGH);
    run = false;
    run2 = true;
  } else if (code == 25573 || code == 3359) { //GDP TEAM ---> START COUNTING ITS TIME
    digitalWrite(rfrele, HIGH);
    digitalWrite(gdprele, LOW);
    run = true;
    run2 = false;
  } else if (code == 29576) { //REFEREE CARD ---> PAUSE TIME
    digitalWrite(rfrele, LOW);
    digitalWrite(gdprele, LOW);
    run = false;
    run2 = false;
  } else if (code == 3829) { //ERASER CARD ---> PAUSE TIME, DELETE IT TO 0
    digitalWrite(rfrele, LOW);
    digitalWrite(gdprele, LOW);
    second = 0;
    minute = 0;
    tenth = 0;
    hour = 0;
    second2 = 0;
    minute2 = 0;
    tenth2 = 0;
    hour2 = 0;
    updateLCD();
    updateLCD2();
    run = false;
    run2 = false;
  }
  tickClock();
}

void tick() {
  if (run) {
    updateLCD(); //zapis cas prveho timu na displej LCD

    if (second == 59) {
      second = 0;
      minute++;

    } else {
      second++;
    }

    if (minute == 60) {
      minute = 0;
      hour++;

    }
  }
}
void tick2() {
  if (run2) {
    updateLCD2(); //zapis cas druheho timu na displej LCD
    if (second2 == 59) {
      second2 = 0;
      minute2++;

    } else {
      second2++;
    }

    if (minute2 == 60) {
      minute2 = 0;
      hour2++;

    }
  }
}
void updateLCD() {
  lcd.setCursor(4, 0);
  if (hour < 10) {
    lcd.print("0");
  }
  lcd.print(hour, DEC);
  lcd.print(":");
  if (minute < 10) {
    lcd.print("0");
  }
  lcd.print(minute, DEC);
  lcd.print(":");

  if (second < 10) {
    lcd.print("0");
  }
  lcd.print(second, DEC);
}
void updateLCD2() {
  lcd.setCursor(4, 1);
  if (hour2 < 10) {
    lcd.print("0");
  }
  lcd.print(hour2, DEC);
  lcd.print(":");
  if (minute2 < 10) {
    lcd.print("0");
  }
  lcd.print(minute2, DEC);
  lcd.print(":");

  if (second2 < 10) {
    lcd.print("0");
  }
  lcd.print(second2, DEC);

}

void dump_byte_array(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {

  }
  code = 10000 * buffer[4] + 1000 * buffer[3] + 100 * buffer[2] + 10 * buffer[1] + buffer[0];
  Serial.println("Card address: ");
  Serial.println(code);
}

martinius96, 2021-03-15:
si, it looks like that statement in loop....
`if (rfid.PICC_IsNewCardPresent() && rfid.PICC_ReadCardSerial()) {`
`...`
`}`
is running 24 up to 26 ms if card is attached or not... It's so long...
And in moment when card is attached it is running 30ms. 
Is there some option how to speed it up a bit? :)

Rotzbua, 2021-03-15:
> [..] running 24 up to 26 ms if card is attached or not... It's so long...

Long? No.

>  Is there some option how to speed it up a bit? :)

The library is simple straight forward. The access to the spi device is blocking. I do not think it is possible to speed it up without changing the fundamental method of the library, e.g. add interrupt usage see #554.

One option without changing the library I see is to use a multicore system like the esp32. The other to use different order and timings.

In general the rfid chip is not designed for timing stuff. It is designed to read and write information to the token.

For timing use light barriers or buttons.

closed 2021-03-15 by Rotzbua