miguelbalboa/rfid #560: Possible workaround for the delay() after `PCD_Init()`

LasseCre, 2021-05-01:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Rotzbua, 2021-05-06:
I updated the readme. I guess this helps more ppl. Best wishes.

LasseCre, 2021-05-06:
Hey, unfortunately my initial post was not quiete right: I made a silly mistake, and instead of soldering the capacitor between GND and 3.3V, I soldered it between GND and Reset. (But hey, it works!) When I realized this I added a 1KOhm resistor to protect the GPIO that is connected to Reset from current spikes.
So it is propably not a general power issue but has something to do with the timing of the reset?
(The Cap should act like a slight "delay" on the Reset pin...)
Sorry I didnt update this earlier!

Rotzbua, 2021-05-06:
I think this needs much testing of hardware and the according lines of code. Maybe someone has the time to test 🧑‍💼 ...

https://github.com/miguelbalboa/rfid/blob/906e57d5e410b7fc2e3756f0712a4c62cbc918a6/src/MFRC522.cpp#L193-L220