miguelbalboa/rfid #564: Library Doesn't Support SoftSPI

kshitijgarg2609, 2021-06-08:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: _____
 * MFRC522 device: _____
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

kshitijgarg2609, 2021-06-08:
@ricardojlrufino , Please can you give the support for SoftSPI of DigitalIO, as I am unable to use other SPI device like SD card adapter along with RFID.

kshitijgarg2609, 2021-06-08:
> @Rotzbua , Please can you give the support for SoftSPI of DigitalIO, as I am unable to use other SPI device like SD card adapter along with RFID.

Rotzbua, 2021-06-08:
Plz use search next time.
Try code of #499

closed 2021-06-08 by Rotzbua