miguelbalboa/rfid #566: Cannot compile for ESP8266

safaenet, 2021-07-13:
### Step 1: Describe your environment

 * OS version: Windows 10 Pro x64
 * Arduino IDE version: 1.8.15
 * MFRC522 Library version: 1.4.8 
 * ESP device: ESP-12E
 * MFRC522 device: RFID-RC522
  
### Step 2: Describe the problem
Cannot compile for ESP8266, But Compiles for Aduino (UNO) without problem.

#### Observed Results:

  * [Error Message.txt](https://github.com/miguelbalboa/rfid/files/6810172/Error.Message.txt)
  
  
#### Relevant Code:

```c++
#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN         9          // Configurable, see typical pin layout above
#define SS_PIN          10         // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

void setup() {
	Serial.begin(9600);		// Initialize serial communications with the PC
	while (!Serial);		// Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
	SPI.begin();			// Init SPI bus
	mfrc522.PCD_Init();		// Init MFRC522
	delay(4);				// Optional delay. Some board do need more time after init to be ready, see Readme
	mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, SAK, type, and data blocks..."));
}

void loop() {
	// Reset the loop if no new card present on the sensor/reader. This saves the entire process when idle.
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}
```

Rotzbua, 2021-07-13:
Not issued by this library.

From log you can see, that you somehow broke your installation. You have multiple configs and also installed this lib twice.
````
Multiple libraries were found for "SPI.h"

 Used: D:\Users\avazeh1\Documents\Arduino\libraries\SPI

 Not used: C:\Users\Safa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\SPI

Multiple libraries were found for "MFRC522.h"

 Used: D:\Users\avazeh1\Documents\Arduino\libraries\MFRC522

 Not used: D:\Users\avazeh1\Documents\Arduino\libraries\rfid-master

closed 2021-07-13 by Rotzbua

safaenet, 2021-07-14:
Any advice how to fix this issue ? i removed all libraries related to RFID/NFC and reinstalled this library again. But it cannot compile for ESP8266. It works fine with Arduino Boards.

Rotzbua, 2021-07-14:
As I mentioned it is a problem with the **Arduino IDE**. Make a complete uninstall and cleanup. For more help please address the Arduino forum/wiki.