miguelbalboa/rfid #575: Issues using the library with an esp32 (Wemos lolin 32 lite)

syntaxbullet, 2021-12-15:
* OS version:  Windows 10 19043
 * Arduino IDE version: using PlatformIO
 * MFRC522 Library version: 1.4.10
 * Arduino device: using Wemos ESP32 lolin lite 
 * MFRC522 device: Generic, authentic MFRC522 version 0x92 / 2.0
  
### Step 2: Describe the problem
Board correctly returns version 2.0 using the firmware dump but fails to read any tags, in any of the examples.
My tags have been shipped together with the reader and confirmed to be MIFARE classic tags using my mobile phone.

#### Steps to reproduce:

#### Observed Results:

  * Dumpinfo example does not return anything / is waiting for a card.
  
#### Expected Results:

  * Info of my card (contact info to be dumped)

vladko312, 2021-12-24:
README file has no pinout for this exact board. What pins were you using for the signals?

Rotzbua, 2021-12-30:
If the example returns a version the wiring should be correct. @syntaxbullet Do you use own code or a specific example?

syntaxbullet, 2021-12-30:
seems like my supplier provided me with some rfid scanners that didn't work, even though the package was new and unopened, ordering a new batch fixed the issue. Thank you for your responses ;)

closed 2021-12-30 by syntaxbullet