miguelbalboa/rfid #579: Timeout in communication.

theLMGN, 2022-01-27:
### Step 1: Describe your environment

 * OS version: macOS 12
 * Arduino IDE version: 1.8.16
 * MFRC522 Library version: eda2e385668163062250526c0e19033247d196a8 
 * Arduino device: Uno
 * MFRC522 device: Generic
  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
  * DumpInfo

#### Steps to reproduce:

  1. Attempt to read NFC tag

  
#### Observed Results:

  ```
Card UID: 69 6A BF 55
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63  PCD_Authenticate() failed: Timeout in communication.
  14     59  PCD_Authenticate() failed: Timeout in communication.
  13     55  PCD_Authenticate() failed: Timeout in communication.
  12     51  PCD_Authenticate() failed: Timeout in communication.
  11     47  PCD_Authenticate() failed: Timeout in communication.
  10     43  PCD_Authenticate() failed: Timeout in communication.
   9     39  PCD_Authenticate() failed: Timeout in communication.
   8     35  PCD_Authenticate() failed: Timeout in communication.
   7     31  PCD_Authenticate() failed: Timeout in communication.
   6     27  PCD_Authenticate() failed: Timeout in communication.
   5     23  PCD_Authenticate() failed: Timeout in communication.
   4     19  PCD_Authenticate() failed: Timeout in communication.
   3     15  PCD_Authenticate() failed: Timeout in communication.
   2     11  PCD_Authenticate() failed: Timeout in communication.
   1      7  PCD_Authenticate() failed: Timeout in communication.
   0      3  PCD_Authenticate() failed: Timeout in communication.

```
  
#### Expected Results:

```
Card UID: XX XX XX XX
Card SAK: 08
PICC type: MIFARE 1KB
Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits
  15     63   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         62   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         61   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         60   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  14     59   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         58   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         57   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
         56   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
  13     55   00 00 00 00  00 00 FF 07  80 69 FF FF  FF FF FF FF  [ 0 0 1 ] 
         54   00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  [ 0 0 0 ] 
```

Same card works in Android phone, same reader works with other card.

Rotzbua, 2022-03-12:
Did you tried an older version?

vladko312, 2022-04-08:
Try using [rfid_default_keys](https://github.com/miguelbalboa/rfid/tree/master/examples/rfid_default_keys) as an example.
A lot of cards are using default keys for sector 0, so that UID can be accessed. Your one uses it as well, since you were able to read UID.
For example, in my city, transport cars are using `A0 A1 A2 A3 A4 A5` as a key for sector 0 (blocks 0-3) and some non-standart keys for the rest, while some other cards I found were using `A0 A1 A2 A3 A4 A5` for all the sectors.
Despite that, [DumpInfo](https://github.com/miguelbalboa/rfid/tree/master/examples/DumpInfo) only dumps bolcks with factory-default keys (`FF FF FF FF FF FF`).