miguelbalboa/rfid #581: Not able to read cards (specifically) unless a card/tag is present while the board starts

moussie369, 2022-03-09:
### Step 1: Describe your environment

 * OS version: W10
 * Arduino IDE version: 1.8.15
 * MFRC522 Library version: 1.4.10
 * Arduino device: nano v3 (clone)
 * MFRC522 device: firmware is 2.0
  
### Step 2: Describe the problem

 Cards won't be read unless one is present when the program starts, if trying to read a card after some times it won't detect anything.
 It works fine when using https://github.com/aiolos/rfid 

#### Affected file(s) or example(s):
  
  *All examples/any code using this library
  *DumpInfo
  
#### Steps to reproduce:

  1. Reset the board
  2. Wait ~10-15s
  3. Place a card in front of the reader
  
#### Observed Results:

 Nothing happens and it won't read anything until the board is reseted. 
 It works fine if only using tags or if a card/tag is present while it's rebooting.
  
#### Expected Results:

  * Detect/dump the card informations to serial

Rotzbua, 2022-05-23:
Did you test `1.4.0`?

moussie369, 2022-05-23:
Yes, 1.3.6 was the last to work

Rotzbua, 2022-05-23:
Can you test current version without connected reset pin and replace code with:
```cpp
MFRC522 mfrc522(SS_PIN, UNUSED_PIN);   // Create MFRC522 instance
```

moussie369, 2022-05-23:
The arduino IDE and platformio won't build with UNUSED_PIN

Rotzbua, 2022-05-23:
Sry, ofc with scope `MFRC522 mfrc522(SS_PIN, MFRC522::UNUSED_PIN);`

moussie369, 2022-05-23:
Well I think I found the culprit, since then I bought a few more nano clones and it seems to work ok with them.
I was kind of freaking out trying to figure out what had changed since march until I realized it.
I still don't get why it's working only with 1.3.6 and earlier, but unplugging rst and putting MFRC522::UNUSED_PIN didn't do anything