miguelbalboa/rfid #583: Having trouble with long wires (+-5m)

podaen, 2022-04-21:
OS version: win7
Arduino IDE version: 1.6/1.8
Arduino Espressif sdk 2.0.2
MFRC522 Library version: 1.4.10
Arduino device: ESP32
MFRC522 device: RFID mini

https://nl.banggood.com/3pcs-RFID-Reader-Module-RC522-Mini-S50-13_56Mhz-6cm-With-Tags-SPI-Write-and-Read-p-1604816.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=BE&utm_content=minha&utm_campaign=minha-beg-nl-pc&currency=EUR&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=sandra&utm_campaign=sandra-ssc-beg-nl-all-0910-13anv&ad_id=381659571753&gclid=Cj0KCQiAip-PBhDVARIsAPP2xc3f3JFAfeX_vbI8ORpCM8l008H68bLvHdPdFX7QN2CC4hxBUsgpKb4aAk6MEALw_wcB

I have two esp32 with the same connections

One with short wiring
One with long wiring

The long wiring doesn't detect the rfid card, the short wiring does with the same shetch. What could be done to resolve this problem?

Before I had it working with sdk 1.0.6. Now I have sdk 2.0.2. In my search to detect the problem I did change anything... Or only changed the reset pin from 21 to 16, but I don't think this is a problem because with the short wiring it works.

```
#include <MFRC522.h> //library responsible for communicating with the module RFID-RC522
//#include <SPI.h> //library responsible for communicating of SPI bus
#include <SPI.h>

#define SS_PIN 5//14//5//10
#define RST_PIN 16//21//27//35//9
#define SIZE_BUFFER 18
#define MAX_SIZE_BLOCK 16
//#define greenPin 12
//#define redPin 32
//used in authentication
MFRC522::MIFARE_Key key;
//authentication return status code
MFRC522::StatusCode status;
// Defined pins to module RC522
MFRC522 mfrc522(SS_PIN, RST_PIN);

/* Set your new UID here! */
#define NEW_UID {0xDE, 0xAD, 0xBE, 0xEF}//old code C2 27 DA 1B , Card SAK: 08 , MIFARE 1KB

//#include "SdFat.h"
//#include "FreeStack.h"
//// **************** SD CARD
//const int SD_CS_PIN = 17;//22;// SS;//test multie spi devices
//#define SPI_SPEED SD_SCK_MHZ(4)
////const char MP3DirPath[] = "23/"; // MP3 path
////const char DirPath[] = "23/"; // MP3 path
//SdFat audio_SD;
//SdFile file;
//SdFile dir;
//int count = 0;

void setup()
{
    Serial.begin(9600);

    pinMode(19, INPUT_PULLUP);//miso needs a pullup (sd card reader)
    pinMode(RST_PIN, OUTPUT);

    SPI.begin(); // Init SPI bus

    //pinMode(greenPin, OUTPUT);
    //pinMode(redPin, OUTPUT);

    // Init MFRC522
    mfrc522.PCD_Init();
    Serial.println("Approach your reader card...");
    Serial.println();

    digitalWrite(RST_PIN, LOW);

[... 205 lines cut ...]

    // Set new UID
    byte newUid[] = NEW_UID;
    if (mfrc522.MIFARE_SetUid(newUid, (byte)4, true)) {
        Serial.println(F("Wrote new UID to card."));
    }

    // Halt PICC and re-select it so DumpToSerial doesn't get confused
    mfrc522.PICC_HaltA();
    if (!mfrc522.PICC_IsNewCardPresent() || !mfrc522.PICC_ReadCardSerial()) {
        return;
    }

    // Dump the new memory contents
    Serial.println(F("New UID and contents:"));
    mfrc522.PICC_DumpToSerial(&(mfrc522.uid));

    //delay(2000);
    readingData();

    Serial.println("Write New UID done");
    while (!true);
    //for (;;) {}//infinet loop
}
```

podaen, 2022-04-21:
It has to be the long wiring, I double checked evrything.

podaen, 2022-04-21:
This is how I connected it

SDA 5
SCK 18
MOSI 23
MISO19
IRQ /
GND GND
RST 16
3V3 3V

podaen, 2022-04-21:
I have it working, The ground was not connected and while testing I have added this line in the setup witch is wrong digitalWrite(RST_PIN, LOW);

starfield33, 2022-04-21:
What I did was to put one ESP32 at each reader, and then let them communicate via wifi,  Bluetooth,  serial or Cbus. You then have ports to open doors & lights. 

 
Very Best Regards, | 
Mit freundlichen Grüßen
Graham Andrews

***@***.***
***@***.***
‭
+44 (0) 7767 867944 Mobile - Europe

podaen, 2022-05-20:
It didn't work aqtually wel. It seems that multiple spi devices aren't really supported
I suggest you add SdSpiConfig spiconfig (CS_PIN, SHARED_SPI, SD_SCK_MHZ(4)); to your library

DrLou, 2022-05-20:
Though we haven’t tested this, empirically, here, I’ve been advised by colleagues expert in SPI that long distance connections shouldn’t be relied upon in designs. 

We use SPI for a number of devices, and had envisioned distancing sensors, for example, at about 10 meters. My sense is that the voltage drops alone would make such a design troublesome.

Lou Picciano

podaen, 2022-05-20:
The length is not really an issue when they work sepratly, but together... I had a simulare issue on keypad with long wiring (non spi), they implemented scantime for that. 

Anyway I had an other project with a display also connected with spi with a length few meters and after a couple of mounths it didn't work anymore ether. I dicided to, to connect them wireless. I think I am going to let it go. It will limit the distance of the rfid to the controller... In my case 5 meters would not be a problem.

I will use esp_now because I am using wifi on that particuare device. And wifi and bluetooth at the same time also doesn't work wel, but thanks for the info.

Rotzbua, 2022-05-21:
As @DrLou mentioned I agree that is is an electrical engineering challenge. There are various reasons for failing. 
I know setups using 2m wiring having stability issues. They used thin not shielded wires. After switching to shielded quality wires it worked.

I do not see an issue in the code of this library. Due to this fact I would recommend @podaen to search help in an electrical engineering forum.

closed 2022-05-21 by Rotzbua

starfield33, 2022-10-11:
Something that i am using with ESP32 is canbus and I’ve got that in multiple ESP32 boards talking to each other and they’re over some good distances. MCP2515 
I’m looking for an ESP32 board with the MCP2515 bus module on it if anyone has seen one let me know. ;) 

 
Very Best Regards, | 
Mit freundlichen Grüßen
Graham Andrews

***@***.***
***@***.***
‭
+44 (0) 7767 867944 Mobile - Europe

podaen, 2022-10-11:
How long do you have it working... Year, month, days? And what protocol, RS485, RS232, ...

podaen, 2022-10-11:
Is still SPI I see, but with a better error handling... Do you have it working with more than two devices? Or is it limited to two devices?

podaen, 2022-10-11:
Is it this board you are using?
[image]