miguelbalboa/rfid #584: printDec() helper function cut and paste error

bhelterline, 2022-06-02:
### Step 1: Describe your environment

 * OS version: _all____
 * Arduino IDE version: _all____
 * MFRC522 Library version: _master-branch____ 
 * Arduino device: __all___
 * MFRC522 device: _all____
  
### Step 2: Describe the problem
The helper function printDec() in the NUID.ino example file has a cut and paste error from the HEX version.  You don't want a leading zero if the value is above 0x10.
/**
 * Helper routine to dump a byte array as dec values to Serial.
 */
void printDec(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], DEC);
  }
}

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

 if the uid happens to be 9, 15, 16, 240, you get the output
"  09 015 16 240"
 
#### Expected Results:

I would expect a 3 digit value for each byte 000 to 255 or at least consistent leading white space
" 009 015 016 240" or
"   9  15  16 240"
  
#### Relevant Code:

```/**
 * Helper routine to dump a byte array as dec values to Serial.
 */
void printDec(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], DEC);
  }
}
```

Rotzbua, 2022-06-04:
@bhelterline Will you create a pull request?

bhelterline, 2022-06-05:
done!

closed 2022-06-06 by Rotzbua