miguelbalboa/rfid #587: Problem with my code or something in IsNewCardPresent()

dageci, 2022-08-17:
### Step 1: Describe your environment

 * OS version: W10 (latest)
 * Arduino IDE version: 1.8.19
 * MFRC522 Library version: 1.4.10
 * Arduino device: ESP32
 * MFRC522 device: MINI RFID - RC522
  
### Step 2: Describe the problem
Sometimes it works and sometimes it doesn't pass the testing of PICC_IsNewCardPresent when I restart the power.

I have added the esp32 led on of to see if it pass this if block:
```c++
   if ( ! mfrc522.PICC_IsNewCardPresent()) {
      return;
   }
   digitalWrite(LED_BUILTIN, HIGH);
   delay(200);
   digitalWrite(LED_BUILTIN, LOW);
```

Sometimes, when it works the led blinks, but sometimes when I switch the power off and then back on, it doesn't pass.

When I put the led blinking inside this if block it is blinking:
```c++
   if ( ! mfrc522.PICC_IsNewCardPresent()) {
      digitalWrite(LED_BUILTIN, HIGH);
      delay(200);
      digitalWrite(LED_BUILTIN, LOW);
      return;
   }
```
  
So I'm not sure if my code is wrong, or is there something else?

#### Relevant Code:

```c++
 #include <SPI.h>
#include <MFRC522.h>
#include "BluetoothSerial.h"

#define SS_PIN 5
#define RST_PIN 22

MFRC522 mfrc522(SS_PIN, RST_PIN);
MFRC522::MIFARE_Key key;

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif
BluetoothSerial SerialBT;

void setup() {
   Serial.begin(115200); 
   SPI.begin();      // Initiate  SPI bus
   mfrc522.PCD_Init();   // Initiate MFRC522

  for (byte i = 0; i < 6; i++) {
    key.keyByte[i] = 0xFF;  
  }

   SerialBT.begin("RFID Card Reader");

  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

}

void loop() {

  
   if ( ! mfrc522.PICC_IsNewCardPresent()) {
      return;
   }
   
   //  just to test if it goes here (blink blue led on esp32)
   digitalWrite(LED_BUILTIN, HIGH);
   delay(200);
   digitalWrite(LED_BUILTIN, LOW);
      

   // Select one of the cards
   if ( ! mfrc522.PICC_ReadCardSerial()) {
      return;
   }

   char str[32] = "";
   array_to_string(mfrc522.uid.uidByte, 4, str); 
   Serial.println(str); 
  SerialBT.print(str);
  SerialBT.println(); 
  delay(1000);  

   mfrc522.PICC_HaltA();

  mfrc522.PCD_StopCrypto1();

}

void array_to_string(byte array[], unsigned int len, char buffer[])
{
   for (unsigned int i = 0; i < len; i++)
   {
      if (i == 0)
        sprintf(buffer, "%02X", array[i]);
      else
        sprintf(&buffer[i*3-1], ":%02X", array[i]);
   }
}
```

dageci, 2022-08-17:
Maybe I have wrongly wired ESP32 & RC522.
In [another forum](https://www.esp8266.com/viewtopic.php?f=6&t=23274&p=92969#p92969) I have find out that I can't use all the GPIO's for RS522.

I'm new to this, if you could take a look if this is ok what I have connected.
I'm using an ESP32 D1 mini

[image]

And my wiring is:

RC522 > ESP32

SDA > I05
SCK > I018
MOSI > I023
MISO > I019
IRQ
GND > GND
RST > I022
3v3 > 3v3

Is that ok?

Rotzbua, 2022-08-30:
You are right. On ESP boards you can not use any pin. There are a few ambiguities and stumbles with the pins.

If the setup works than all good 👍 

A recommendable overview is from [Andreas Spiess](https://www.youtube.com/c/AndreasSpiess) on YT:
https://www.youtube.com/watch?v=LY-1DHTxRAk

closed 2022-08-30 by Rotzbua