miguelbalboa/rfid #601: Compile error with Arduino ESP32 release 2.0.8

tueddy, 2023-04-23:
Hi,

with the new Arduino ESP32 release 2.0.8 this library does not longer compiles due to using `__FlashStringHelper `

For reference see
https://github.com/espressif/arduino-esp32/issues/8108
https://github.com/espressif/arduino-esp32/pull/7941

Any way to get arround this bug (feature)?

dok-net, 2023-04-23:
> Any way to get arround this bug (feature)?

Yes: https://github.com/espressif/arduino-esp32/pull/8111

paulocoutinhox, 2023-04-29:
Hi,

I have the same problem:
https://forum.arduino.cc/t/macro-f-error/1118126/8

```
The implementation of the F() macro was changed in ESP32 core version 2.0.8.
It is now

#define F(string_literal) (string_literal)
instead of casting to const __FlashStrinHelper *.
This breaks use cases like

const __FlashStringHelper *MFRC522::GetStatusCodeName(MFRC522::StatusCode code  ///< One of the StatusCode enums.
                                        ) {
    switch (code) {
        case STATUS_OK:             return F("Success.");
        case STATUS_ERROR:          return F("Error in communication.");
        ...
        default:                    return F("Unknown error");
    }
} // End GetStatusCodeName()
```

Until the solution, downgrade to ESP32 core 2.0.7.

tueddy, 2023-04-29:
Until the problem is solved in this library you can use my fork
https://github.com/tueddy/rfid.git

Or change "__FlashStringHelper " to "char" and remove "F()", see changes here: https://github.com/miguelbalboa/rfid/compare/master...tueddy:rfid:master

jerryg2003, 2023-05-03:
Had same problem.  Change F(“X”) in “return” statements in MFRC522.cpp to “((__FlashStringHelper *) “X”)” and all works fine with ESP32 2.0.8.

tueddy, 2023-05-23:
Problem is fixed with Arduino 2.0.9 (PlatformIO Package 6.3.0)

closed 2023-05-23 by tueddy