miguelbalboa/rfid #603: RC522 Mini Not Work vs RC522 Normal

ruielias, 2023-06-19:
### Step 1: Describe your environment

 * OS version: _____
 * Arduino IDE version: _____
 * MFRC522 Library version: _____ 
 * Arduino device: ESP32-WROOM-32D
 * MFRC522 devices: MW-ET LIVE MINI RFID-RC522 and RFID-RC522
  
### Step 2:

Hello,
I have a full size RC522 module and working 100% OK
I paid for 3 "MW-ET LIVE MINI RFID-RC522" units, and none of them work, but they pass the "PCD_PerformSelfTest()".
Are Minis different?

#### Affected file(s) or example(s):
 
 
#### Steps to reproduce:
  
#### Observed Results:

  * Normal Size RC522 :

MFRC522 Digital self test
Firmware Version: 0x92 = v2.0
Only known versions supported
Performing test...
Result: OK

  * MW-ET LIVE MINI RFID-RC522 :

MFRC522 Digital self test
Firmware Version: 0x92 = v2.0
Only known versions supported
Performing test...
Result: OK
  
#### Expected Results:

  * Read on both RC522 types. But I can only read TagID in normal format
  
#### Relevant Code:

```c++
 #include <Arduino.h>
#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN  16
#define RST_PIN 13
 
MFRC522 rfid(SS_PIN, RST_PIN);

void setup() {
  Serial.begin(115200);
  while (!Serial);

  SPI.begin(5,19,27,16);
  rfid.PCD_Init(); // init MFRC522
  delay(10);
  Serial.println("Iniciou RFid");
  
  delay(100);

 Serial.println(F("*****************************"));
  Serial.println(F("MFRC522 Digital self test"));
  Serial.println(F("*****************************"));
  rfid.PCD_DumpVersionToSerial();  // Show version of PCD - MFRC522 Card Reader
  Serial.println(F("-----------------------------"));
  Serial.println(F("Only known versions supported"));
  Serial.println(F("-----------------------------"));
  Serial.println(F("Performing test..."));
  bool result = rfid.PCD_PerformSelfTest(); // perform the test
  Serial.println(F("-----------------------------"));
  Serial.print(F("Result: "));
  if (result)
    Serial.println(F("OK"));
  else
    Serial.println(F("DEFECT or UNKNOWN"));
  Serial.println();
}

void loop() {
  if (rfid.PICC_IsNewCardPresent()) { // new tag is available
    Serial.println("NEW...");
    if (rfid.PICC_ReadCardSerial()) { // NUID has been readed
      MFRC522::PICC_Type piccType = rfid.PICC_GetType(rfid.uid.sak);
      Serial.print("RFID/NFC Tag Type: ");
      Serial.println(rfid.PICC_GetTypeName(piccType));
      // print UID in Serial Monitor in the hex format
      Serial.print("UID:");
      for (int i = 0; i < rfid.uid.size; i++) {
        Serial.print(rfid.uid.uidByte[i] < 0x10 ? " 0" : " ");
        Serial.print(rfid.uid.uidByte[i], HEX);
      }
      Serial.println();
      rfid.PICC_HaltA(); // halt PICC
      rfid.PCD_StopCrypto1(); // stop encryption on PCD
    }
  }
  delay(250);
}
```
[image]
[image]

alex-eri, 2023-06-26:
it works if pulldown RST pin. see #599

alastaira, 2023-09-16:
Without seeing the schematic, it sounds like they may have omitted the pullup resistor and/or put an inverter on the RST line. Normally, this pin can be ignored - it's pulled HIGH by default, and you only need write a LOW signal to it to reset the reader. From the description given by @alex-eri , it sounds like you need to do the opposite for this board.

alex-eri, 2023-11-28:
Not inverted. Pull down needed to make this line work 
https://github.com/miguelbalboa/rfid/blob/1e73cd34d977c5fe17fa5f2e2e8174650a572028/src/MFRC522.cpp#L210C2-L210C2

alex-eri, 2023-11-28:
pulled HIGH by default only on board without FLOATING input.

phillipkusinski, 2024-04-10:
Hey guys, I am having the exact same issue. I'm using the ESP32 DIY More and my wiring looks like this:
3V3=3V3
RST=G16
GND=GND
MISO=G19
MOSI=G23
SCK=G18
SDA=G15

My code looks like this (its just for testing):
`#include <SPI.h>
#include <MFRC522.h>

#define SS_PIN 15  // SDA Pin
#define RST_PIN 16  // Reset Pin 

MFRC522 mfrc522(SS_PIN, RST_PIN); // Create MFRC522 instance

void setup() {
  pinMode(RST_PIN, OUTPUT); // Set RST pin as output
  digitalWrite(RST_PIN, LOW); // Set RST pin to LOW to ensure proper initialization
  Serial.begin(9600); // Initialize serial communications
  SPI.begin();        // Init SPI bus
  mfrc522.PCD_Init(); // Init MFRC522
  Serial.println("Scan for RFID tags...");
}

void loop() {
  // Look for new cards
  if (mfrc522.PICC_IsNewCardPresent() && mfrc522.PICC_ReadCardSerial()) {
    Serial.println("RFID-Tag erkannt!");
    delay(1000); // Verzögerung, um wiederholte Ausgabe zu vermeiden
  }
}
`

It would be nice to get some help:)

tbcse, 2025-02-21:
> Without seeing the schematic, it sounds like they may have omitted the pullup resistor

This is correct. The the reset point on the black colored MINI RFID-RC522 is wired directly to the chip. Unlike the standard blue colored RFID-RC522 board, it does not have a pullup resistor. Adding a 10K resistor between RST and 3V3 solved the reset issue and the random operability when powered on.

There's also at least 2 models of the Mini RFID-RC522 on a green PCB, I do not know if those have the same hardware difference.

ndombaedwin017-arch, 2026-05-09:
Scan PICC to see UID, SAK, type, and data blocks... but not work