miguelbalboa/rfid #604: How to get 5 bytes UID ?

msw1998, 2023-07-25:
I want to get 5th byte output from MIFARE Classic Card instead of 4 bytes output.
Currently I am getting this UID from arduino code 
8B 65 66 B9
but from the raspberry pi code of SimpleMFRC522.py I get this uid in hex 
8B 65 66 B9 31 

I can't update the raspberry pi code to change the output from 5 byte to 4 byte as it is already in production 
Is there anyway I can get the 5th byte from this library?

Rotzbua, 2023-07-28:
Block 0 contains the raw data for uid.

closed 2023-07-28 by Rotzbua

msw1998, 2023-07-28:
How can I print the raw data for the uid?

I did run the dumpinfo and it only gives me 4 bytes as mentioned above 
i tried increase uid->size to 5 and increase the bytesToCopy variable to 5 as well 
then it outputted the 5th byte as 00