miguelbalboa/rfid #605: Help i can't get the RC522 module for RFID to work.

Manuelmine1, 2023-07-25:
Hello i wanted to ask a question i can't get the RC522 module for RFID to work.
At first I tried first through The DumpInfo example is what I get in the terminal:
"Firmware Version: 0x18 = (unknown)
"Scan PICC to see UID, SAK, type, and data blocks...
"But by approaching an rfid tag it is not read.

Then I tried with The Firmware_check example and the result is this:
"MFRC522 Digital self test
"*****************************
"Firmware Version: 0x18 = (unknown)
"-----------------------------
"Only known versions supported
"-----------------------------
"Performing test...
"-----------------------------
"Result: DEFECT or UNKNOWN

Could the error be from incompatible firmware?
Thank you very much

OS version: Win10
Arduino IDE version: 2.1.1
MFRC522 Library version: 1.4.10 
Arduino device: Arduino uno R3
MFRC522 device: RFID-RC522

--------
[DumpInfo.zip](https://github.com/miguelbalboa/rfid/files/12165223/DumpInfo.zip)
[firmware_check.zip](https://github.com/miguelbalboa/rfid/files/12165228/firmware_check.zip)

[image]

alastaira, 2023-09-11:
Seeing the _other_ end of those wires would really help.

alastaira, 2023-09-16:
> Hi! Has the problem been solved? I had the same problem.

The problem is, most likely, that you've got a poor connection on your MISO line, which is corrupting the response from the reader. How have you got it wired?

jisle15064468204, 2023-10-13:
> > Hi! Has the problem been solved? I had the same problem.
> 
> The problem is, most likely, that you've got a poor connection on your MISO line, which is corrupting the response from the reader. How have you got it wired?
I have resolved that the slice selection (CS) GPIO pins on my development board are not supported. Replaced with a GPIO solution.