miguelbalboa/rfid #632: "ordered comparison of pointer with integer zero" error ArduinoIDE / ESP32

CnczubehoerEu, 2024-06-06:
### Step 1: Describe your environment

 * OS version: Windows 11
 * Arduino IDE version: 2.3.2
 * MFRC522 Library version: 1.4.11 
 * Arduino device: DOIT ESP32 DEVKIT V1
 * MFRC522 device: MFRC522 (??)
  
### Step 2: Describe the problem
My ESP32 with MFRC522 worked well for a very long time. I put my work on hold for a few months and wanted to get back to it today. I pressed "verify" on the since the last work few month ago code and now the following error appears:

XYZ\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp: In member function 'MFRC522::StatusCode MFRC522Extended::TCL_Transceive(TagInfo*, byte*, byte, byte*, byte*)':
XYZ\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:824:34: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
  824 |         if (backData && (backLen > 0)) {
      |                          ~~~~~~~~^~~
XYZ\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:847:42: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
  847 |                 if (backData && (backLen > 0)) {
      |                                  ~~~~~~~~^~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\WiFi
  Not used: C:\Users\XXX\OneDrive\Dokumente\Arduino\libraries\WiFi
exit status 1

Compilation error: exit status 1

I did not yet connect the board to the USB port. If I do so, the same error comes. I de- and reinstalled the MFRC522 library but without success. 

#### Affected file(s) or example(s):
  
  * Filename 1
  * Filename 2
  
#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, screenshot, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

I reduced my code to a minium: 

```c++
#include <MFRC522.h>

const int PIN_RFID_SDA =     21;
const int PIN_RFID_RST =     22;

MFRC522 mfrc522(PIN_RFID_SDA, PIN_RFID_RST);

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

Any idea, what could I try further? Thank you very much in advanced for your support!

Gustavo-daCosta, 2024-06-09:
Same problem here.

OS version: Windows 11
Arduino IDE version: 2.3.2
MFRC522 Library version: 1.4.11
Arduino device: ESP32-WROOM-32
MFRC522 device: RFID-RC522

CnczubehoerEu, 2024-06-09:
Ich changes the following manually in MFRC522Extended.cpp:

Line 824: if (backData && backLen && (*backLen > 0)) {
Line 847: if (backData && backLen && (*backLen > 0)) {

No it works for me.

However, still I do not understand, what's changed in the last few month...

Gustavo-daCosta, 2024-06-09:
Yeah, changed manually the code and now it passes, but my RFID module still doesnt recognize any RFID cards.

Probably an hardware problem

Anyway, thank you.

CnczubehoerEu, 2024-06-09:
In my case, I can read everything I want. I guess, in your case it is indeed an second problem somewhere else.

Jan21021, 2024-06-09:
Same problem here. Problem is with a NodeMCU Esp 32s. When i use the ESP32 Boardmanager 2.0.17 the problem does not exist. The Problem only exists with the newer ESP32 board packs like 3.0 or 3.0.1. 

Hope that helps a little bit to narrow down the problem.. =)

Mer08, 2024-06-10:
I think this is happening because of the new IDF release

Jeff86, 2024-06-16:
Hey, I had the same issue, I fixed it modifying with `if (backData && (backLen != nullptr))` instead of `if (backData && (backLen > 0))` for both occurrences. It works great on ESP32 now and I also tested on ESP8266.

Gustavo-daCosta, 2024-06-18:
The pull request #625 solves this issue

But it seems that the repository stoped his development, so any pull request are being approved and merged.

So the alternatives are creating a fork with the updates, or use an existing fork

PCBuil, 2025-01-31:
i cant find if (backData && (backLen > 0)) {
for both of the lines