miguelbalboa/rfid #639: When compiling the code with the MFRC522 library, I encountered an error in `MFRC522Extended.cpp`. The error involves an ordered comparison of a pointer with an integer.

vidura2, 2024-10-05:
OS version: Windows 11
Arduino IDE version: 2.3.3
MFRC522 Library version: 1.4.11
Arduino device: ESP32
MFRC522 device: RC522 RFID Module

Affected file(s) or example(s):

    MFRC522Extended.cpp

Steps to reproduce:

    Add the MFRC522 library to the Arduino project.
    Compile a program using the ESP32 and RC522 RFID Module.
    Encounter the compilation error.

ERROR:
 e:\Arduino IDE\libraries\MFRC522\src\MFRC522Extended.cpp: In member function 'MFRC522::StatusCode MFRC522Extended::TCL_Transceive(TagInfo*, byte*, byte, byte*, byte*)':
e:\Arduino IDE\libraries\MFRC522\src\MFRC522Extended.cpp:824:34: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
824 |         if (backData && (backLen > 0)) {
    |                          ~~~~~~~~^~~
e:\Arduino IDE\libraries\MFRC522\src\MFRC522Extended.cpp:847:42: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
847 |                 if (backData && (backLen > 0)) {
    |                                  ~~~~~~~~^~~

The code :

#include <WiFi.h>
#include <SPI.h>
#include <MFRC522.h>
#include <time.h>
#include <AESLib.h> // Library for AES encryption

// WiFi credentials (if needed, currently commented out)
// const char* ssid = "ESP32_Access_Point";
// const char* password = "123456789";

// RFID setup
#define SS_PIN    5   // CS/SDA pin for RFID
#define RST_PIN   22  // RST pin for RFID
#define RELAY_PIN 25  // Relay control pin

MFRC522 rfid(SS_PIN, RST_PIN);

// Authorized RFID UIDs (replace these with your own values)
byte managerKeyUID[4]   = {0x3A, 0xC9, 0x6A, 0xCB};
byte secretaryKeyUID[4] = {0x30, 0x01, 0x8B, 0x15};

// Rolling code setup
byte rollingCode[4] = {0x00, 0x00, 0x00, 0x01}; // Initial rolling code
byte validRollingCode[16]; // Encrypted rolling code

// AES encryption key and initialization vector (IV)
byte aesKey[16] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F};
byte aesIV[16] = {0x00};

// WebServer for access log
#include <WebServer.h>
WebServer server(80);

// Log for RFID access attempts
String loggedEntries = "";
String systemStatus = "";

// Time synchronization
void setupTime() {
    configTime(0, 0, "pool.ntp.org");
    setenv("TZ", "UTC-5", 1); // Adjust timezone to your location
    tzset();

    struct tm timeinfo;
    int retryCount = 0;
    while (!getLocalTime(&timeinfo) && retryCount < 5) {
        Serial.println("Failed to sync time, retrying...");
        delay(1000);
        retryCount++;
    }

    if (retryCount == 5) {
        systemStatus = "Failed to sync time, using default time.";
    } else {
        systemStatus = "Time synchronized.";
    }
}

// Get the current time as a string
String getTime() {
    struct tm timeinfo;
    if (!getLocalTime(&timeinfo)) {
        return "Failed to obtain time";
    }
    char timeStr[50];
    strftime(timeStr, sizeof(timeStr), "%Y-%m-%d %H:%M:%S", &timeinfo);
    return String(timeStr);
}

// Serve login page
void handleLoginPage() {
    String html = "<html><head><style>";
    html += "body { font-family: Arial; background-color: #f0f0f0; margin: 0; padding: 20px; }";
    html += "h1 { color: #333; }";
    html += "table { width: 100%; border-collapse: collapse; margin-top: 20px; }";
    html += "th, td { border: 1px solid #ddd; padding: 8px; text-align: left; }";
    html += "th { background-color: #4CAF50; color: white; }";
    html += "tr:nth-child(even) { background-color: #f2f2f2; }";
    html += "</style></head><body>";
    html += "<h1>RFID Door Lock System</h1>";
    html += "<p>System Status: " + systemStatus + "</p>";
    html += "<table>";
    html += "<tr><th>UID</th><th>Date</th><th>Time</th></tr>";
    html += loggedEntries;
    html += "</table></body></html>";

    server.send(200, "text/html", html);
}

// Log RFID entry with UID and timestamp
void logEntry(byte uid[]) {
    String uidStr = "";
    for (int i = 0; i < 4; i++) {
        uidStr += String(uid[i], HEX);
        if (i < 3) uidStr += ":";
    }

    String dateTime = getTime();
    String date = dateTime.substring(0, 10); // Extract date (YYYY-MM-DD)
    String time = dateTime.substring(11);    // Extract time (HH:MM:SS)

    String log = "<tr><td>" + uidStr + "</td><td>" + date + "</td><td>" + time + "</td></tr>";
    loggedEntries += log;
}

// Check if RFID UID is authorized
bool isAuthorizedUID(byte uid[], byte authorizedUID[]) {
    for (int i = 0; i < 4; i++) {
        if (uid[i] != authorizedUID[i]) {
            return false;
        }
    }
    return true;
}

// Encrypt rolling code using AES
void encryptRollingCode(byte code[], byte encryptedCode[]) {
    AESLib aes;
    aes.encrypt(code, 16, encryptedCode, aesKey, 128, aesIV);
}

// Decrypt rolling code using AES
void decryptRollingCode(byte encryptedCode[], byte decryptedCode[]) {
    AESLib aes;
    aes.decrypt(encryptedCode, 16, decryptedCode, aesKey, 128, aesIV);
}

// Generate new rolling code
void generateNewRollingCode() {
    for (int i = 0; i < 4; i++) {
        rollingCode[i] = (rollingCode[i] + random(1, 255)) % 255;
    }
    encryptRollingCode(rollingCode, validRollingCode);
}

// Check if rolling code is valid
bool isValidRollingCode(byte currentCode[]) {
    byte decryptedCode[16];
    decryptRollingCode(validRollingCode, decryptedCode);

    for (int i = 0; i < 4; i++) {
        if (currentCode[i] != decryptedCode[i]) {
            return false;
        }
    }
    return true;
}

void setup() {
    Serial.begin(115200);

    // Start WiFi access point (optional)
    // WiFi.softAP(ssid, password);
    // systemStatus = "Access Point started. IP Address: " + WiFi.softAPIP().toString();

    // Initialize RFID
    SPI.begin();
    rfid.PCD_Init();
    pinMode(RELAY_PIN, OUTPUT);
    digitalWrite(RELAY_PIN, HIGH); // Lock the door by default

    // Start web server for log display
    server.on("/", handleLoginPage);
    server.begin();

    // Setup time synchronization
    setupTime();

    // Initialize rolling code encryption
    encryptRollingCode(rollingCode, validRollingCode);
}

void loop() {
    server.handleClient(); // Handle web server requests

    // Check for RFID card/tag
    if (rfid.PICC_IsNewCardPresent() && rfid.PICC_ReadCardSerial()) {
        if (isAuthorizedUID(rfid.uid.uidByte, managerKeyUID) || isAuthorizedUID(rfid.uid.uidByte, secretaryKeyUID)) {
            
            // Validate the rolling code
            if (isValidRollingCode(rfid.uid.uidByte)) {
                logEntry(rfid.uid.uidByte); // Log the access
                digitalWrite(RELAY_PIN, LOW); // Unlock door
                delay(2000); // Keep door unlocked for 2 seconds
                digitalWrite(RELAY_PIN, HIGH); // Lock door

                // Generate new rolling code
                generateNewRollingCode();
                systemStatus = "Access granted and rolling code updated.";
            } else {
                systemStatus = "Invalid rolling code!";
            }
        } else {
            systemStatus = "Unauthorized access attempt.";
        }

        // Halt RFID reader
        rfid.PICC_HaltA();
        rfid.PCD_StopCrypto1();
    }
}

gcandrade10, 2024-10-10:
Same error here

vidura2, 2024-10-11:
added a pull request

frivera26, 2024-10-29:
El problema se resuelve de la siguiente manera:

 if (backData && (*backLen > 0)) {

y ya puedo usar el RFID-RC522

Jsuryaboi-08, 2024-11-07:
same error please help.

Jsuryaboi-08, 2024-11-07:
guys I have found the solution for this problem.

Open MFRC522Extended.cpp in a text editor. It should be located in your Arduino libraries folder:

c:\Users\[username]\OneDrive\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp

_Locate lines 824 and 847 with the following code:_

**if (backData && (backLen > 0)) {**

_Modify both lines to remove the invalid pointer comparison by replacing them with:_

**if (backData != nullptr && backLen != nullptr && *backLen > 0) {**

The original code incorrectly compares a pointer (backLen) with an integer (0). This modification uses != nullptr to check if backData and backLen pointers are valid and then dereferences backLen to compare its value with 0, fixing the error.

robosphere99, 2025-01-12:
I have solved in the PR

Bobootk, 2025-02-03:
i had the same problem 
I did this solution and it works
many thanks for the solution

robosphere99, 2025-02-17:
I have solved it in the PR