miguelbalboa/rfid #646: MFRC522Extended.cpp:824:34: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')

0101prashant, 2024-12-28:
### Step 1: Describe your environment
 * OS version: Win11
 * Arduino IDE version: 2.3.2
 * MFRC522 Library version: 1.4.11
 * Arduino device: ESP32 

  
### Step 2: Describe the problem

#### Affected file(s) or example(s):
  
MFRC522Extended.cpp
  
#### Steps to reproduce:

#### Observed Results:

  libraries\MFRC522\src\MFRC522Extended.cpp:824:34: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
  824 |         if (backData && (backLen > 0)) { 
  
\MFRC522Extended.cpp:847:42: error: ordered comparison of pointer with integer zero ('byte*' {aka 'unsigned char*'} and 'int')
  847 |                 if (backData && (backLen > 0)) {
  
#### Expected Results:

  it should compile without error
  
#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

mr-rezaaa, 2025-01-06:
I have same problem

meshya, 2025-01-06:
@0101prashant, @mr-rezaaa I fixed this bug.
Check https://github.com/meshya/rfid-bugfix

0101prashant, 2025-01-06:
> @0101prashant, @mr-rezaaa Fixed here https://github.com/meshya/rfid-bugfix Waiting for merge 
[we both waitining to get accepted pull] ( https://github.com/0101prashant/rfid) Waiting for merge  was in MFRC522Extended.cpp if condition i already fixed it i was working with esp32 that day.

closed 2025-01-06 by 0101prashant