miguelbalboa/rfid #649: Hi, I'm new here!!! I'm not able to read RFID tag with my two arduino UNO with the module RC522 RFID.

jordannguerin, 2025-01-14:
Hello, I bought two Arduinos to test my RFID RC522. I installed the MFRC522 library by github communauty, and every time I start the program with my two arduinos, it tells me this error: **WARNING: Communication failure, is the MFRC522 properly connected? Scan PICC to see UID, SAK, type and data blocks...**.  I tried attaching all the cables directly to my arduino and also tried with a breadboard, but every time it said that. I've watched a lot of videos, but nothing works. Can anyone help me please???

dyzin, 2025-01-14:
Hey Jordan,

I had the same issue and couldn't figure out why. It seems to be a problem with the library. Try using the fork https://github.com/vidura2/MFRC522 instead. That fixed the problem for me.

AMfaraz, 2025-01-28:
I was going to write try connecting directly without breadboard but you already done it, If you are 100% sure jumper are connect to correct cables on both arduino and mfrc522 then try changing the module.